project-chip/connectedhomeip #24992: Address review comments from PR 24986

bzbarsky-apple, 2023-02-11:
Review comments there were not addressed before it was merged.  See https://github.com/project-chip/connectedhomeip/pull/24986#discussion_r1103481942

Specific actions:

1. Undo the changes to the certification tests that were not accompanied by test plan changes.
2. Add the test that was asked for in the review comments.

@jrhees-cae @yufengwangca

bzbarsky-apple, 2023-02-11:
Of course ideally the test plans would in fact be modified so they catch this issue and then the test would be one of the certification tests.... @cjandhyala

cjandhyala, 2023-02-14:
@jrhees-cae Could you update the test plans, if the expected outcome in a TC is modified , we must execute the TC in a TE or SVE.

jrhees-cae, 2023-02-14:
@cjandhyala I looked at the test plan document related to this test case (DRLK-10) and there aren't any needed changes. The test plan does not specifically call out what UserIndex or CredentialIndex to use for the test (the only thing I changed was the credentialIndex used during the steps).

jrhees-cae, 2023-02-14:
> @jrhees-cae <https://github.com/jrhees-cae> Ok, I see verification steps
> are updated to verify certain expected values, since the expected outcome
> is changed in some of the steps in the script, will need to run this script
> in a TE/SVE.
>
> I see the script is updated to verify the below values.
>
>    1. CredentialIndex: 1 is changed to CredentialIndex: 2
>    2. verify "UserIndex and Credentials are set to null"
>
> Also I see that expected outcome in one of the step is 'DataIndex is set
> to 4' but in the actual verification that value is not verified. Is that
> fine ?

Yes that is fine. It does not need to be verified.

bzbarsky-apple, 2023-02-22:
Fixed by https://github.com/project-chip/connectedhomeip/pull/25030

PR #25030 (2023-02-14): Add tests in DL_LockUnlock yaml to validate Lock/Unlock credentials and userIndex (#24992)
Add tests in DL_LockUnlock yaml to:
- verify correct CredentialIndex on LockOperation events
- verify NULL UserIndex and Credentials on LockOperation errors when using invalid PINCodes

Fixes #24992
commit d3d7df97 (2023-02-14): Add tests in DL_LockUnlock yaml to validate Lock/Unlock credentials and userIndex (#24992) (#25030)

* Add tests in DL_LockUnlock yaml: (#24992)
- verify correct CredentialIndex on LockOperation events
- verify NULL UserIndex and Credentials on LockOperation errors when using invalid PINCodes
Fixes #24992

* Restyled by prettier-yaml

* Exclude DL_LockUnlock yaml test from Darwin (waiting on readEvent support)

---------
closed 2023-02-22 by bzbarsky-apple