project-chip/connectedhomeip #25001: [XML] Update Time Format Localization Cluster XML to match updated spec

ReneJosefsen, 2023-02-11:
The data types names has been updated in Time Format Localization Cluster and must be updated in the XML as well.

**Updated**
TimeZoneType -> TimeZoneStruct
DstOffsetType  -> DSTOffsetStruct

Spec PR: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6184 (already merged)

bzbarsky-apple, 2023-02-13:
Change missing from the above list: the `DstOffset` attribute got renamed to `DSTOffset`.

bzbarsky-apple, 2023-02-13:
And some of the events got renamed, but the SDK does not contain those so far anyway.

ReneJosefsen, 2023-02-14:
You are right, I was so focused on the data types that I completely forgot the impact of the Dst to DST change. I have updated the description.

closed 2023-02-14 by woody-apple