project-chip/connectedhomeip #25002: [1.0]

cyyyys, 2023-02-12:
### Reproduction steps

ask a very basic question...
how to generate "app/PluginApplicationCallbacks.h" need in util.cpp

### Bug prevalence

not a bug

### GitHub hash of the SDK that was being used

tag 1.0.0.2

### Platform

freeRTOS

### Type

Platform Issue

cyyyys, 2023-02-12:
[image]

cyyyys, 2023-02-12:
[image]

bzbarsky-apple, 2023-02-12:
@andy31415

andy31415, 2023-02-13:
The file is compile-time generated by `codegen.py` based on input .matter files.

Specifically the build rules in:

`src/app/chip_data_model.gni` or `src/app/chip_data_model.cmake` contain the rules to create them (which in turn call rules from `build/chip/chip_codegen.gni` or  `build/chip/chip_codegen.cmake`.

The invocation generally calls `codegen.py` with a `generator` and `output-dir` argument for a specific matter file. Rules are in https://github.com/project-chip/connectedhomeip/blob/master/build/chip/chip_codegen.gni#L39, but maybe not as readable. Here is an example call generating things:

```
> source scripts/activate.sh
> ./scripts/codegen.py --generator cpp-app --output-dir out/genfiles examples/lighting-app/lighting-common/lighting-app.matter
2023-02-13 10:03:56 INFO    Parsing idl from examples/lighting-app/lighting-common/lighting-app.matter
2023-02-13 10:03:56 INFO    Running code generator cpp-app
2023-02-13 10:03:56 INFO    File to be generated: app/PluginApplicationCallbacks.h
2023-02-13 10:03:56 INFO    Creating output directory: out/genfiles/app
2023-02-13 10:03:56 INFO    Writing new data to: out/genfiles/app/PluginApplicationCallbacks.h
2023-02-13 10:03:56 INFO    File to be generated: app/callback-stub.cpp
2023-02-13 10:03:56 INFO    Writing new data to: out/genfiles/app/callback-stub.cpp
2023-02-13 10:03:56 INFO    Done
> ls out/genfiles/app/
callback-stub.cpp  PluginApplicationCallbacks.h
```

andy31415, 2023-02-13:
@cyyyys - this information is obviously not easily discoverable ... would some details in https://github.com/project-chip/connectedhomeip/blob/master/docs/code_generation.md have helped?

I would like to write some description of this, would like to place it and formulate it in some way that would have been useful for you. What would have helped?

closed 2023-02-13 by andy31415