project-chip/connectedhomeip #25003: [BUG] Field order doesn't match between spec and code

nicolas17, 2023-02-12:
(Sorry for not using the issue form but none seemed to be directly applicable to this issue)

In the Matter Core Specification 1.0 (22-22349-001_Matter-1.0-Core-Specification.pdf), section 4.4 table 8, it says protocol messages have a Protocol ID field followed by an optional Protocol Vendor ID field (depending on whether the V flag is set).

However, [PayloadHeader::Decode](https://github.com/project-chip/connectedhomeip/blob/bbfd5d6275f508f1e7779385a2ad974930cf9cfb/src/transport/raw/MessageHeader.cpp#L261) and [PayloadHeader::Encode](https://github.com/project-chip/connectedhomeip/blob/bbfd5d6275f508f1e7779385a2ad974930cf9cfb/src/transport/raw/MessageHeader.cpp#L377) in src/transport/raw/MessageHeader.cpp (as of bbfd5d6275f5) are reading/writing the Vendor ID first and *then* the Protocol ID. Which one is correct?

bzbarsky-apple, 2023-02-12:
Oh, nice catch....

The SDK is correct by definition at this point.  It looks like the spec _used_ to have:
```
|4 bytes 2+|Message Protocol Id
```
and that was later changed to:
```
|2 bytes 2+|Message Protocol Id
|2 bytes 2+| [ Message Protocol Vendor Id ]
```
in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/401.  I guess the assumption was that a little-endian 4-byte (vendor id, protocol id) pair would have the vendor id in the high bytes, so would have the protocol id first.

The 4-byte protocol id was there ever since https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/22

At the same time, the implementation until https://github.com/project-chip/connectedhomeip/pull/1950 was missing this stuff entirely and after that was changed to:
```
*  16 bit: | Optional Vendor ID                                                   |
*  16 bit: | Protocol ID                                                          |
```
which happened 10 days _before_ the first addition of this stuff to the spec...  So it's not even that the implementation was "wrong" at the point.  And no one has changed the implementation since then.

I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6275 to get the spec fixed.

@turon

nicolas17, 2023-02-16:
Thanks for the quick response! I appreciate the history digging, although I don't have access to the spec repo :)

bzbarsky-apple, 2023-02-16:
This has been fixed in the spec with https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6288

closed 2023-02-16 by bzbarsky-apple