project-chip/connectedhomeip #25012: [Documentation]  Explain difference between bridge-app and dynamic-bridge-app

svigelj, 2023-02-13:
### Documentation issues

What is the actual difference between bridge-app and dynamic-bridge-app and how should a manufacturer choose one over the other to start with their development?

They both seem to do the same thing (dynamically adding and removing devices), but dynamic-bridge-app seems a bit more complex while not all functionality works with it out of the box (eg. adding BridgedInformationCluster to `bridge-app.zap` breaks the build).

It would be great if further explanation would be provided on the goals of each of the two examples. Is one just a legacy example? Are there actually different purposes for each of them?

I apologize if this is not an appropriate issue format, but as a manufacturers we are fairly confused which of the two examples to choose to base our future development on.
...

bzbarsky-apple, 2023-02-13:
@dnicoara

achaulk-goog, 2023-02-15:
The plan was to replace bridge app entirely with dynamic bridge, however the bridge app had several existing behaviours directly coded that looked like test cases (press a key, add a predefined light) that dynamic bridge wouldn't support as easily, so it was decided to rename it and land in parallel

The idea with the dynamic bridge was to be able to support all clusters in any configuration via code generation and runtime creation of endpoints, whereas the original bridge app has plumbing code for each cluster type and declares device types using the dynamic macros but still has all possible device configurations declared directly in the sources

svigelj, 2023-02-16:
@achaulk-goog thank you very much for better clarifying the difference.

If I then understand this correctly, both examples can be used as a base to build any kind of dynamic bridging app. The difference is that dynamic-bridge-app example will have more building blocks already prepared (with all possible options for clusters being pre-generated via ZAP) , while bridge-app is more of a "build it yourself" example (you need to define all the clusters you need with macros). Do I understand this correctly?  Thanks again.

andy31415, 2023-08-24:
dynamic-bridge-app was removed.

closed 2023-08-24 by andy31415