project-chip/connectedhomeip #25018: Building Matter fails on MacOS.  Does not handle environment path file names with spaces.

kskenyon, 2023-02-13:
### Reproduction steps

Build normally.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

?

### Platform

darwin

### Anything else?

Build fails because script won't set pathway properly.  Source directory is /Users/private/Visual Studio Codes

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (7m31.9s)
  Setting up Python environment.....done (7m43.9s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

cat: /Users/private/Visual: No such file or directory
cat: Studio: No such file or directory
cat: Codes/connectedhomeip/.environment/env_root.txt: No such file or directory
 
Can't build examples because script won't set pathway properly.  Source directory is /Users/private/Visual Studio Codes

  ACTIVATOR! This sets your shell environment variables.

cat: /Users/private/Visual: No such file or directory
cat: Studio: No such file or directory
cat: Codes/connectedhomeip/.environment/env_root.txt: No such file or directory
➜

bzbarsky-apple, 2023-02-13:
Is this basically a duplicate of `https://github.com/project-chip/connectedhomeip/issues/23420`?  @mspang 

This does not look MacOS-specific; it's a problem in the bootstrap machinery....

mspang, 2023-02-13:
https://gn-review.git.corp.google.com/c/gn/+/15080

mspang, 2023-02-13:
I don't know if this is the same bug. It may be.

kskenyon, 2023-02-13:
I put it into a different directory to build but the space problem still exists.

kskenyon, 2023-02-13:
Didn't find #23420 because of the not helpful naming.

bzbarsky-apple, 2023-02-13:
@kskenyon Yes, I only knew the other thing existed because I saw it before; expecting you to find it would not be reasonable...