project-chip/connectedhomeip #25023: [BUG] Segmentation fault on ExchangeContext::StartResponseTimer

domichae-amazon, 2023-02-13:
### Reproduction steps

### Problem

When an `ExchangeContext` is `ShutDown()` while the `ExchangeContext` is in the middle of sending a message (`SendMessage`), the internal state of the `ExchangeContext` is torn down resulting in a segmentation fault on accessing `GetSessionManager()->SystemLayer()`.

### Impact

The `Darwin` version of the TV casting example crashes. This theoretically could apply to any example, though, as the `ExchangeContext` is a core component.

### Bug prevalence

Sometimes

### GitHub hash of the SDK that was being used

02bc67e245dbc0e85b3ab772c1f81eb27b953420

### Platform

darwin

### Platform Version(s)

N/A

bzbarsky-apple, 2023-02-13:
> When an ExchangeContext is ShutDown() while the ExchangeContext is in the middle of sending a message (SendMessage)

I'm not sure what you mean by "ShutDown()" (there is no such API on ExchangeContext), but what exact thing nulled out `mExchangeMgr` and how did it get called in the middle of `SendMessage`?

Or is the real issue that `SendMessage` is being called after the _exchange manager_ has been shut down?  @domichae-amazon

domichae-amazon, 2023-02-13:
That's correct - the issue is that the Exchange Manager was shut down, which in turn clears out some of the state of the Exchange Context which may still be alive.

bzbarsky-apple, 2023-02-13:
> the issue is that the Exchange Manager was shut down, which in turn clears out some of the state of the Exchange Context which may still be alive.

Why is the exchange context still alive?  All exchanges should be closed before the exchange manager shuts down, precisely because of this problem.  Who is not doing that, and why not?

What is the _exact_ sequence of events that got us here?

domichae-amazon, 2023-02-17:
We discussed this issue more over on the pull request. I did some more digging based on the information that @bzbarsky-apple shared and found the root cause: the `darwin` variant of the `tv-casting-app` example project does not take the Matter stack lock before any interactions with the Matter SDK. This doesn't match the expectations of the SDK and could theoretically result in crashes in a few different places. The specific event leading to a crash here is that the sample iOS app calls `Server::GetInstance().Shutdown()` in a non-thread-safe way (from the application's main DispatchQueue). I'll update the content of the bug to indicate the real issue and create a fresh pull request that addresses it.