project-chip/connectedhomeip #25033: [BUG] CHIPTool crashes when running the readLocal Temperature command for a thermostat cluster.

HirokiOgawa1129, 2023-02-13:
### Reproduction steps

1.Press the PROVISION CHIP DEVICE WITH THREAD button in the CHIPTool app to commission the thermostat device.
2.Press the CLUSTER INTERACTION TOOL button on the CHIPTool app to transition to the screen for accessing the thermostat cluster.
3.Execute the following command.
[image]
4.An error occurs as below.
[image]

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/master/examples/android/CHIPTool

### Platform

android

### Platform Version(s)

Android ver 13

### Anything else?

The app also crashes when I run WriteOccupiedCoolingSetpoint .

joonhaengHeo, 2023-02-15:
This issue appears to be caused by missing some Cluster Callback classes when create ClusterInfo.java using ZAP.

yunhanw-google, 2023-02-15:
@joonhaengHeo we miss some generated cluster callback? or this issue is caused by new cluster with missing generated things?

joonhaengHeo, 2023-02-16:
@yunhanw-google 
https://github.com/project-chip/connectedhomeip/blob/83dd188990e57ab141e2a07672ca55e0962367c5/src/controller/java/zap-generated/chip/devicecontroller/ClusterReadMapping.java#L10179

It seems that an issue occurred in the line above. This is where the Java Type Casting Exception occurs, and some ZAP files need to be modified.

The ZAP file has been modified and the commit has been uploaded. #25098

PR #25098 (2023-02-21): [Android] Fix Cluster Interaction issue
Fixes #25033 
1. Modify ZAP file related cluster interaction.
2. Add null check in CHIPTool android.
closed 2023-02-21 by andy31415