project-chip/connectedhomeip #25040: Use separate transitions for hue and saturation in color control

bzbarsky-apple, 2023-02-14:
Right now, a move on the hue will terminate an in-progress move on the saturation, and vice versa.

This is pretty unfortunate, from a client's point of view, and has no basis in the spec as far as I can tell.  We should probably consider just doing separate transitions for the two (but allow the MoveToHueAndSaturation/etc commands to combine into a single transition).  

One possible approach: we could support doing up to two transitions per endpoint, where each transition is transitioning hue, or saturation, or both.  When a new transition wants to start, there are three cases:

1. It wants to transition both; cancel any existing transitions.
2. It wants to transition hue only.  Cancel any existing hue-only transition, convert any existing hue-and-saturation transition to saturation-only.
3. It wants to transition saturation only.  Cancel any existing saturation-only transition, convert any existing hue-and-saturation transition to hue-only.

@jmartinez-silabs thoughts?

jmartinez-silabs, 2023-02-14:
@bzbarsky-apple I agree, separating the two gives a better user experience and more flexibility and I don't see any cons either. It is fairly straightforward to implement. 

Note that hue and enhanced hue will be grouped together has the same transition.

bzbarsky-apple, 2023-02-14:
Yep, there's only one hue to transition, so enhanced hue and hue have to use the same thing.

PR #25345 (2023-03-09): [ColorControl] Split (Enhanced)Hue and Saturation transitions
fixes #25040

Hue and Saturation are part of the same color schema (HSV). Previously, a command changing only the hue would stop any active saturation transition and vice versa.

This PR reworks the use of the ColorHueTransitionState and SaturationTransitionState so each transition can continue unaffected by a command only targeted to change one of the two.
Examples
```
move-to-hue
move-saturation
enhanced-step-hue
etc.
```

The following commands affect both at the same time:
```
move-to-hue-and-saturation
enhanced-move-to-hue-and- [... cut]
closed 2023-03-09 by jmartinez-silabs