project-chip/connectedhomeip #25085: [ICD] Refactor subscription timer logic to send all updates at the same time

mkardous-silabs, 2023-02-15:
### Feature description

For ICDs, it is preferable to wake up once to send all updates at the same time instead of waking up multiple times. The subscription logic needs to be refactored to allow for cross-fabric synchronization.

### Platform

all

### Platform Version(s)

1.1

lpbeliveau-silabs, 2023-07-20:
The logic is implemented and integrated by:
https://github.com/project-chip/connectedhomeip/pull/27943
https://github.com/project-chip/connectedhomeip/pull/28104

However, the crossfabric synchronisation fails the TestReadInteraction unit test as this test does not expect this behavior.
An equivalent test needs to be put in place to validate our current synchronisation behaviour and preserve it in the event of modifications to the ReportScheduler.

turon, 2023-08-28:
WG: Is the TestReadInteraction failure due to implementation error or test design error?

@jmartinez-silabs to follow up on details

lpbeliveau-silabs, 2023-08-28:
@turon Test design error, TestReadInteraction has test cases where multiple read handlers are expected to generate different reports at different times.  For instance, `TestSubscribeUrgentWildcardEvent` expects an urgent report to be emitted at the end of the minimal interval and checks that a non urgent report isn't emitted. The syncrhonized behavior would fail this test.

Granted, it is currently disabled for icd devices but we are still not testing the syncrhonized behavior in any test.

turon, 2023-08-29:
I'm assuming our design emits the emergency event at the min interval, but bundles in the non emergency information as well and that extra info is triggering the fail?  How difficult will it be for the certification/test team to relax the pass criteria and update the harness for this?

lpbeliveau-silabs, 2024-01-02:
Update : TestReadInteraction now supports the Synchronized Report Scheduler since: https://github.com/project-chip/connectedhomeip/pull/30608 was merged.

Currently we have https://github.com/project-chip/connectedhomeip/pull/31134 documenting the logic behind the synchronisation scheduling that needs to be review. We should close this issue once the documenting PR is merged unless a change in the synchronisation logic is deemed necessary.