project-chip/connectedhomeip #25100: [Documentation] How to implement matter audio products？

Vieufoux, 2023-02-16:
### Documentation issues

Hi, I want to develop and manufacture matter audio products. I have some questions below:

1. I want develop a matter wireless speaker that has the features of play, pause, adjusting volume, previous track, next track, audio input selection, audio output selection...... How to use [Matter SDK](https://github.com/project-chip/connectedhomeip) to implement? I can build chip-tools and linux-x64-all-clusters app in repo now , but I don't know how to implement audio product using SDK? I found some clusters in media device type, but I don't know how to use these clusters to implement my products.
2. I want develop a matter amplifier that has the feature of adjusting gain, adjusting DSP parameters...... This is not a standard device type in Matter document. How can i implement it? and if I implement a custom device type, does it mean that other ecosystems(Alexa, Siri, Google Nest Hub...) can't control my product?

### Platform

IMX8

bzbarsky-apple, 2023-02-16:
> I want develop a matter wireless speaker that has the features of play, pause, adjusting volume, previous track, next track, audio input selection, audio output selection

There's no such device type defined in the device library right now; you would probably need to get that to happen...  You could perhaps pretend like it's a video player with no video, but previous/next track are not things anywhere in there.

> does it mean that other ecosystems(Alexa, Siri, Google Nest Hub...) can't control my product?

Correct, unless they have some sort of out-of-band knowledge of what your non-standard clusters are...

Vieufoux, 2023-02-16:
@bzbarsky-apple 
Thanks for your quick reply.

We are an audio product manufacturer. According to what you said, we are unable to develop and manufacture a fully featured audio products that support matter at current stage because no such device types.

If we want Matter to add speakers and amplifiers device types, could you tell me what are the ways to achieve? If our company joins CSA members, is it possible to allow us to propose corresponding standards? Or submit an application to expedite the development of this standard?

andy31415, 2023-02-16:
https://csa-iot.org/become-member/

There is some description about membership at a glance, including things like: `Participant Members contribute to and develop standards they will later adopt and use. Participants can access Alliance Working Groups, are hands-on in developing specifications, and gain early access to draft specifications and testing, with opportunity to develop and get to market faster.`