project-chip/connectedhomeip #25102: [1.0]

cyyyys, 2023-02-16:
### Reproduction steps

The code has a problem in src/lib/dnssd/minimal_mdns/AddressPolicy.cpp.

line 30. 

VerifyOrDie(gAddressPolicy != nullptr);

Normal here gAddressPolicy should not equal nullptr.

But the value of gAddressPolicy is nullptr here, and the  program is stuck here.

So the bluetooth could not be found next...

### Bug prevalence

not a bug

### GitHub hash of the SDK that was being used

tag 1.0.0.2

### Platform

freeRTOS

### Type

Platform Issue

bzbarsky-apple, 2023-02-16:
@andy31415

cyyyys, 2023-02-16:
I guess the cause of the problem is that some CPP files are not introduced in Make.defs.

cyyyys, 2023-02-16:
[image]

cyyyys, 2023-02-16:
The path of the select.h in my project is /usr/include/x86_64-linux-gnu/sys/select.h

cyyyys, 2023-02-16:
I solved this problem.

in src/lib/dnssd/MinimalMdnsServer.cpp

open the CHIP_MINMDNS_DEFAULT_POLICY macro

closed 2023-02-16 by cyyyys

andy31415, 2023-02-16:
@cyyyys for future bug reports, please provide more context on things. Here it looks like you are using some custom compilation instructions and such info would be useful in the bug report.