project-chip/connectedhomeip #25105: [Feature] DoorLockServer: Add source attribute for lock/unlock commands

deveritec-rosc, 2023-02-16:
### Feature description

In the current implementation of the door-lock server, `emberAfPluginDoorLockOnDoorLockCommand` and `emberAfPluginDoorLockOnDoorUnlockCommand` do not expose the source of the lock/unlock command.

In the example (and a current PoC I am working on), this information would be beneficial.

From the current interface, I see no easy way to add this functionality, since it would either
- break the function by adding the source argument at the correct place (before the return parameter), or
- breaks the used style that the `out`-parameter is placed last

### Platform

all

bzbarsky-apple, 2023-02-16:
@deveritec-rosc I'm not sure what "break the function" means.  Both options would "break" the function in the sense that it would need to be changed in existing implementations.

I think fundamentally we have two options: we can either change the signature and all consumers have to notice and update (probably any basic testing would catch this need), or we add a second function we call and then implementations that want the extra info implement that, not the existing one.

deveritec-rosc, 2023-02-17:
Yeah, I get it. I was just thinking that adding an optional paramter to the header may solve this, but the function would be recognized as an overloaded one.

From my point of view, changing the whole function signature and hoping that basic testing (or compile time errors) may catch this appears to be the "cleaner" solution. Adding a second function introduces the need for some static checks to avoid unintended behavior, but I would be open for that.

bzbarsky-apple, 2023-02-17:
@deveritec-rosc Compile time errors will not catch this, since this function is defined as a weak symbol.  But yes, basic testing should.

bzbarsky-apple, 2023-10-16:
@deveritec-rosc What "source" information were you looking for?  The current `emberAfPluginDoorLockOnDoorLockCommand` and `emberAfPluginDoorLockOnDoorUnlockCommand` signatures seem to include a fabric index and node ID (when those are available).