project-chip/connectedhomeip #25137: Apple Home Thread Accessory

djandrew2005, 2023-02-16:
### Reproduction steps

It is not possible to add a matter over thread accessory to Apple home without a HomeHub with Thread. Apple should be use other Thread Border Router to communicate with the accessory through them. An iPad should be able to control a matter-over-thread through a nest hub(Thread Border Router)

### Platform

darwin

### Type

Platform validated

bzbarsky-apple, 2023-02-17:
@djandrew2005 This has nothing to do with Thread.  Apple Home requires a HomeHub for Matter devices, whether they are Thread or WiFi.

closed 2023-02-17 by bzbarsky-apple

djandrew2005, 2023-02-17:
But an iPad is a home hub according to this support document and it supports matter accessories. https://support.apple.com/en-us/HT207057

bzbarsky-apple, 2023-02-17:
Hmm.  It looks like this is supposed to work for WiFi (because the iPad knows the WiFi credentials), but for thread we don't know those unless the BR communicates them to us.

@djandrew2005 This is the wrong place in general for Home issues.  What you probably want to do is follow instructions below to file a feedback assistant ticket.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

djandrew2005, 2023-02-17:
I have already submitted feedback for a lot of matter problems including this one but there are no fixes to be found in the new updates. That's why I posted one of them in here

bzbarsky-apple, 2023-02-18:
@djandrew2005 The problem is that Apple's internal code is not tracked here, so filing issues about that code here is not very useful either....

If you do file the feedback ticket for this, though, please post the ticket number here and I will see what I can do to get it followed up on.  But again, note that it really depends on the exact border router used and on whether it plays nice in terms of telling the iPad the Thread network credentials....

djandrew2005, 2023-02-18:
I can understand that.

Ticket number: FB11994480

bzbarsky-apple, 2023-02-22:
@djandrew2005 Thank you!  One thing to check: does the Google Nest Hub actually have a thread network turned on when you are trying to do the pairing?

djandrew2005, 2023-02-22:
> @djandrew2005 Thank you!  One thing to check: does the Google Nest Hub actually have a thread network turned on when you are trying to do the pairing?

I'm using eve door sensor and it's  already paired to Google Home app.  Nest Hub uses Thread to communicate with the sensor. Bluetooth is disabled after the pairing is complete.

bzbarsky-apple, 2023-02-22:
Hmm.  So the sensor is already on-network?  And you open a commissioning window on it, but then the iPad can't commission it?

djandrew2005, 2023-02-22:
Yes. I turn on pairing from Google Home app and I try to pair it with Apple Home. iPad shows an error messaging stating that a Thread Border Router is required for pairing.

djandrew2005, 2023-03-05:
I tried to add the accessory to Home Assistant but I saw the same error. I managed to do it using and android device. So it is possible. Apple needs to fix this! I also updated the support request with more info

bzbarsky-apple, 2023-03-06:
@djandrew2005 Yes, thank you.  For the on-network case there should be no requirement for Thread credentials, and if Home is requiring that it needs to be fixed.