project-chip/connectedhomeip #25148: [1.0] ios home matter no working

cheng5230, 2023-02-17:
### Reproduction steps

**1. connectedhomeip\examples\light-switch-app\esp32 project**
**2  ios connect   show no button config** 
       ios version 16.3.2
[image]

### Bug prevalence

app 

### GitHub hash of the SDK that was being used

13f5e8b012d76b06eb4918b3a618615253fc151f

### Platform

esp32

### Platform Version(s)

1.0.1

### Type

Common Cluster Logic

bzbarsky-apple, 2023-02-17:
@cheng5230 Can you please follow instructions below to file a feedback assistant ticket?  Then put the ticket number in this issue.  Please make sure to attach a sysdiagnose from your resident (AppleTV or HomePod) to the ticket.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

woody-apple, 2023-06-23:
Closing, please follow up via feedback assistant here.

closed 2023-06-23 by woody-apple