project-chip/connectedhomeip #25151: Reading custom cluster from Android App

falconeta, 2023-02-17:
Hi, I developing an Android app for control our devices with Matter. I know that, for custom data, is possible to create a custom cluster for write custom attributes. 
Now, i need to read this data from the Android App but the ChipCluster wrapper doesn't expose any custom cluster methods to call.

Is possible to read a custom cluster's attribute with Android? Using the chiptool cli I able to read this data with the "any"  command.

joonhaengHeo, 2023-02-20:
If you know EndpointID, ClusterID, and AttributeID, I think attribute read is possible through the UI in the Wildcard Button.

joonhaengHeo, 2023-02-21:
@falconeta 
I can't check the detailed log, so it's hard to answer correctly. However, if the part you said to check with string is to check through the method (getValue) below, it may be caused by the failure to parse the TLV.
https://github.com/project-chip/connectedhomeip/blob/e64736c9c151b45653581f749d585bddd9da1ff1/src/controller/java/src/chip/devicecontroller/model/AttributeState.java#L42

The value of getTlv method, which is byteArray before parsing, also needs to be checked.
https://github.com/project-chip/connectedhomeip/blob/e64736c9c151b45653581f749d585bddd9da1ff1/src/controller/java/src/chip/devicecontroller/model/AttributeState.java#L49

falconeta, 2023-02-22:
Hi @joonhaengHeo, after further investigation, i found that the DecodeAttributeValue method returns the error CHIP_ERROR_IM_MALFORMED_ATTRIBUTE_PATH_IB if the clusterId isn't recognized. So, in not possible to read a custom ID (example 0x131bfc00) ?

Logs

> I/ViewRootImpl@7f659f0[CHIPToolActivity]: ViewPostIme pointer 0
I/ViewRootImpl@7f659f0[CHIPToolActivity]: ViewPostIme pointer 1
D/CSM: FindOrEstablishSession: PeerId = [1:0000000000000001]
D/CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
D/DIS: Found an existing secure session to [1:0000000000000001]!
D/DIS: OperationalSessionSetup[1:0000000000000001]: State change 1 --> 5
D/ChipClient: Got connected device pointer
D/DMG: SendReadRequest ReadClient[0x71b0c91000]: Sending Read Request
D/EM: <<< [E:53879i S:17283 M:18585606] (S) Msg TX to 1:0000000000000001 [D22F] --- Type 0001:02 (IM:ReadRequest)
D/IN: (S) Sending msg 18585606 on secure session with LSID: 17283
D/DMG: MoveToState ReadClient[0x71b0c91000]: Moving to [AwaitingIn]
W/libEGL: EGLNativeWindowType 0x71b6718010 disconnect failed
D/OpenGLRenderer: endAllActiveAnimators on 0x71ae77a900 (RippleDrawable) with handle 0x71aecc05c0
D/ViewRootImpl@7f659f0[CHIPToolActivity]: dispatchDetachedFromWindow: reset blurmask
I/ViewRootImpl@7f659f0[CHIPToolActivity]: dispatchDetachedFromWindow
I/SurfaceControl: nativeRelease nativeObject s[488217701920]
I/SurfaceControl: nativeRelease nativeObject e[488217701920]
D/InputTransport: Input channel destroyed: 'd8405ba', fd=102
I/SurfaceControl: nativeRelease nativeObject s[488219645856]
I/SurfaceControl: nativeRelease nativeObject e[488219645856]
I/SurfaceControl: nativeRelease nativeObject s[488219651136]
I/SurfaceControl: nativeRelease nativeObject e[488219651136]
I/SurfaceControl: nativeRelease nativeObject s[488219651040]
I/SurfaceControl: nativeRelease nativeObject e[488219651040]
I/ViewRootImpl@91aa2c6[CHIPToolActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
D/InputMethodManager: prepareNavigationBarInfo() DecorView@896cf3[CHIPToolActivity]
D/InputMethodManager: getNavigationBarColor() -855310
D/InputMethodManager: prepareNavigationBarInfo() DecorView@896cf3[CHIPToolActivity]
D/InputMethodManager: getNavigationBarColor() -855310
V/InputMethodManager: Starting input: tba=com.google.chip.chiptool ic=com.android.internal.widget.EditableInputConnection@684674e mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
D/InputMethodManager: startInputInner - Id : 0
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
D/InputTransport: Input channel constructed: 'ClientS', fd=94
D/InputTransport: Input channel destroyed: 'ClientS', fd=114
I/SurfaceControl: nativeRelease nativeObject s[488376574336]
I/SurfaceControl: nativeRelease nativeObject e[488376574336]
I/SurfaceControl: nativeRelease nativeObject s[488376574048]
I/SurfaceControl: nativeRelease nativeObject e[488376574048]
I/SurfaceControl: nativeRelease nativeObject s[488219646048]
I/SurfaceControl: nativeRelease nativeObject e[488219646048]
D/EM: Retransmitting MessageCounter:18585606 on exchange 53879i Send Cnt 1
D/IN: (S) Sending msg 18585606 on secure session with LSID: 17283
D/EM: >>> [E:53879i S:17283 M:248784996 (Ack:18585606)] (S) Msg RX from 1:0000000000000001 [D22F] --- Type 0001:05 (IM:ReportData)
D/EM: Found matching exchange: 53879i, Delegate: 0x71b0c91010
D/EM: Rxd Ack; Removing MessageCounter:18585606 from Retrans Table on exchange 53879i
D/DMG: ReportDataMessage =
D/DMG: {
D/DMG: 	AttributeReportIBs =
D/DMG: 	[
D/DMG: 		AttributeReportIB =
D/DMG: 		{
D/DMG: 			AttributeDataIB =
D/DMG: 			{
D/DMG: 				DataVersion = 0x15961dd1,
D/DMG: 				AttributePathIB =
D/DMG: 				{
D/DMG: 					Endpoint = 0x0,
D/DMG: 					Cluster = 0x131bfc00,
D/DMG: 					Attribute = 0x0000_0000,
D/DMG: 				}
D/DMG: 					
D/DMG: 				Data = false, 
D/DMG: 			},
D/DMG: 			
D/DMG: 		},
D/DMG: 		
D/DMG: 	],
D/DMG: 	
D/DMG: 	SuppressResponse = true, 
D/DMG: 	InteractionModelRevision = 1
D/DMG: }
D/EM: inside OnAttributeData callback
D/EM: inside OnAttributeData callback 2
D/EM: inside OnAttributeData callback 3
D/EM: inside OnAttributeData callback 4
D/EM: inside OnAttributeData callback 5
D/EM: inside OnAttributeData callback 6
D/EM: decode 320601088
E/-: error type: ../../src/controller/java/zap-generated/CHIPAttributeTLVValueDecoder.cpp:21681: CHIP Error 0x000000B5: Malformed Interacton Model Attribute Path IB
D/EM: inside OnAttributeData callback 6 1 1
I/WildcardFragment: Received wildcard report
I/WildcardFragment: 
I/WildcardFragment: wildcard report Done
D/EM: <<< [E:53879i S:17283 M:18585607 (Ack:248784996)] (S) Msg TX to 1:0000000000000001 [D22F] --- Type 0000:10 (SecureChannel:StandaloneAck)
D/IN: (S) Sending msg 18585607 on secure session with LSID: 17283
D/EM: Flushed pending ack for MessageCounter:248784996 on exchange 53879i
D/IN: Received a duplicate message with MessageCounter:248784996 on exchange 53879i
D/EM: >>> [E:53879i S:17283 M:248784996 (Ack:18585606)] (S) Msg RX from 1:0000000000000001 [D22F] --- Type 0001:05 (IM:ReportData)
D/EM: Generating StandaloneAck via exchange: 53879i
D/EM: Forcing tx of solitary ack for duplicate MessageCounter:248784996 on exchange 53879i
D/EM: <<< [E:53879i S:17283 M:18585608 (Ack:248784996)] (S) Msg TX to 1:0000000000000001 [D22F] --- Type 0000:10 (SecureChannel:StandaloneAck)
D/IN: (S) Sending msg 18585608 on secure session with LSID: 17283
D/EM: >>> [E:53879i S:17283 M:248784997 (Ack:18585606)] (S) Msg RX from 1:0000000000000001 [D22F] --- Type 0000:10 (SecureChannel:StandaloneAck)
E/EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:305: CHIP Error 0x00000070: Unsolicited msg with originator bit clear

joonhaengHeo, 2023-02-23:
@falconeta As I said before, it is not being decoded because it is a custom cluster.
Therefore, it seems that after receiving byteArray using getTlv(), Parsing should be performed at the app end.

falconeta, 2023-02-23:
@bzbarsky-apple sorry, but i don't understand where i parse the data. 
Actually I call the readPath on Android layer with a return callback, in this callback the result, if i pass the custom id, is an empty Map.

https://github.com/project-chip/connectedhomeip/blob/e64736c9c151b45653581f749d585bddd9da1ff1/src/controller/java/AndroidCallbacks.cpp#L239

this method return the error CHIP_ERROR_IM_MALFORMED_ATTRIBUTE_PATH_IB and return empty Map.

So, the AttributeState is instantiated after the TLV validation and i don't know where retrieve this class for call the getTlv method.

this is the method that i use for get the data:

https://github.com/project-chip/connectedhomeip/blob/7e728bc1bd0275f1dff8d75b2214c745047f6e01/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/WildcardFragment.kt#L159-L182

falconeta, 2023-02-23:
> @falconeta I'm not sure why you're asking me... I don't know anything about the Android CHIPTool, really

Sorry, i wrong mentioning

joonhaengHeo, 2023-02-24:
@falconeta 
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/zap-generated/CHIPAttributeTLVValueDecoder.cpp

TLV is parsing for each attribute in each cluster in the file above. If there is no Parsing here, the CHIP_ERROR_IM_MALFORMED_ATTRIBUTE_PATH_IB Error will be returned as you said.
This file is based on the content defined in the ZAP file.
Custom Clusters are not parsed because they are not added to the ZAP.

As I analyzed the code, TLV ByteArray is not being delivered either. When CHIP_ERROR_IM_MALFORMED_ATTRIBUTE_PATH_IB Error occurs, it is shutting down immediately. Even if Parsing is not possible, it would be better if the TLV value is added and corrected.

zzz2016, 2023-02-27:
For now, we have to modify the source code, like ZAP files, and run the specific command to regenerate the codes.

zzz2016, 2023-02-27:
btw, the android sdk is really hard to use as many features are not yet supported.
Is there a plan or roadmap to fix the issues?

falconeta, 2023-02-27:
@zzz2016 do you know if is available a guide for add a cluster? Which files needs to be modified for generate the "zap-generated" files with the custom cluster?

amprazak, 2023-05-19:
Is there any guide or tutorial on how to add a custom cluster

yunhanw-google, 2023-07-27:
Hi @falconeta 
With the latest code, I think you should be able to use android chip-tool to read custom cluster,
You should be able to bypass CHIP_ERROR_IM_MALFORMED_ATTRIBUTE_PATH_IB error.
You could use latest generic Java/JNI API to play around tlv byteArray, and quickly validate your custom cluster for command/read/write.

yunhanw-google, 2023-07-27:
@zzz2016 feel free to add feature request in this project https://github.com/orgs/project-chip/projects/10, yes, we do improve android chip-tool as much as we can

closed 2023-11-20 by falconeta