project-chip/connectedhomeip #25190: [1.0] ninja -C out/host

cyyyys, 2023-02-20:
### Reproduction steps

I met an error when I excute "ninja -C out/host " to construct chip-tool

ninja: error: '../../third_party/nlunit-test/repo/src/nlunit-test.c', needed by 'obj/third_party/nlunit-test/repo/src/libnlunit-test.nlunit-test.c.o', missing and no known rule to make it

### Bug prevalence

not a bug

### GitHub hash of the SDK that was being used

tag 1.0.0.2

### Platform

freeRTOS

### Type

Test Improvement

bzbarsky-apple, 2023-02-20:
@cyyyys Do you have local changes to that submodule?  `third_party/nlunit-test/repo/src/nlunit-test.c` should exist if the submodule is inited at all.

closed 2023-02-21 by cyyyys

kaiser-ren, 2023-09-14:
I have exact issue on Raspberry Pi4 Ubuntu Server, how did you solve this issue?

MandeepTekmar, 2023-11-09:
I am facing the same issue. There is no src folder under repo for nlunit-test. Please help.