project-chip/connectedhomeip #25195: pigweed zap.app has broken symlinks

bzbarsky-apple, 2023-02-20:
Trying to run zap.app from the pigweed-installed bits does not work.  It crashes like so:
```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/Electron Framework.framework/Electron Framework
...
Reason: tried: '.environment/cipd/packages/zap/zap.app/Contents/Frameworks/Electron Framework.framework/Electron Framework' (not a mach-o file)
```

What I see is that:

1. `.environment/cipd/packages/zap/zap.app/Contents/Frameworks/Electron Framework.framework/Electron Framework` instead of being a symlink like it's supposed to be is a normal file, containing the text "Versions/Current/Electron Framework".
2. `.environment/cipd/packages/zap/zap.app/Contents/Frameworks/Electron Framework.framework/Versions/Current` instead of being a symlink like it's supposed to be is a normal file, containing the text "A".
3. `.environment/cipd/packages/zap/zap.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework` exists, and is the right thing: it's a mach-o file, with both arm64 and x86_64 bits in it.

So this would presumably work if the symlinks had not been corrupted.

tpudlik, 2023-02-22:
We've root-caused this (Python's `zipfile`, used in generating the CIPD package, does not by default preserve symlinks), and are working on a fix.

tpudlik, 2023-02-23:
Proposed fix: https://chromium-review.googlesource.com/c/infra/luci/recipes-py/+/4285739.

bzbarsky-apple, 2023-04-06:
@tpudlik https://chromium-review.googlesource.com/c/infra/luci/recipes-py/+/4285739 looks like it's merged, but the symlinks in the cipd version of ZAP are still corrupted?

tpudlik, 2023-04-06:
Which OS do you observe this issue on? Is this Mac, Windows, or Linux? And are you at the tip of the main branch?

On Mac, the CIPD package version currently referred to from the main branch (`2@v2023.03.30-nightly.1`) appears to have working symlinks.

mspang, 2023-04-06:
This would be on Mac.

@bzbarsky-apple , it looks good to me..

```
spang-macbookpro2:zap.app spang$ pwd
/Users/spang/connectedhomeip/.environment/cipd/packages/zap/zap.app
spang-macbookpro2:zap.app spang$ ls -l Contents/Frameworks/Electron\ Framework.framework/
total 0
lrwxr-xr-x  1 spang  primarygroup   35  6 Apr 14:54 Electron Framework -> Versions/Current/Electron Framework
lrwxr-xr-x  1 spang  primarygroup   24  6 Apr 14:54 Helpers -> Versions/Current/Helpers
lrwxr-xr-x  1 spang  primarygroup   26  6 Apr 14:54 Libraries -> Versions/Current/Libraries
lrwxr-xr-x  1 spang  primarygroup   26  6 Apr 14:54 Resources -> Versions/Current/Resources
drwxr-xr-x  4 spang  primarygroup  128  6 Apr 14:54 Versions
spang-macbookpro2:zap.app spang$ ./Contents/MacOS/zap 
🔧 Using state directory: /Users/spang/.zap
```

bzbarsky-apple, 2023-04-06:
I'm on Mac.  But I might have an old-ish version, looks like.  Mine are all dated March 13.  Let me try re-bootstrapping.

bzbarsky-apple, 2023-04-06:
Yes, after re-bootstrap this looks fixed.  Thank you!

closed 2023-04-06 by bzbarsky-apple