project-chip/connectedhomeip #25196: [BUG] command not found: pw

maxkondor, 2023-02-20:
### Reproduction steps

I'm trying to install Matter on my machine (Manjaro XFCE). Doing it step by step:
1. Install pre-requirements
2. Clone repo recursively.
3. Run scripts/bootstrap.sh. All is OK.
4. Run scripts/activate.sh. All is OK.
5. Trying to build esp32 based example (I already have esp-idf4.4.4) Having an error: 
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/max/Tools/matter/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from idl.matter_idl_parser import CreateParser
  File "/home/max/Tools/matter/connectedhomeip/scripts/idl/matter_idl_parser.py", line 7, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
```

6. Trying to run `deactivate` than `bootstrap` again, then have an error: `/home/max/Tools/matter/connectedhomeip/.environment/activate.sh:210: command not found: pw`

A week ago I on version 1.0.0.1 all works fine. 

### Bug prevalence

4 days

### GitHub hash of the SDK that was being used

none

### Platform

esp32

### Platform Version(s)

1.0.0.2

bzbarsky-apple, 2023-02-20:
@maxkondor See the updated build instructions on tip.  The ones on 1.0.0.2 do not really work for esp32...

But briefly, you need to run `activate.sh` after you run the ESP32 `export.sh`...

maxkondor, 2023-02-21:
@bzbarsky-apple yep, you are right. In 1.0.0.2 you must earlier activate esp32 environment and then Matter environment. To fix that error about pw (broken path ???) remove the .environment folder activate esp32 env then activate Matter.

closed 2023-02-21 by maxkondor