project-chip/connectedhomeip #25205: Dose commissioning over Wi-Fi on Apple Homepod mini or Apple TV?

livebase, 2023-02-21:
Dose commissioning over Wi-Fi on Apple Homepod mini or Apple TV?
Does Matter work on Apple Homopod mini or Apple TV if the device does not have BLE, only Wi-Fi?

bzbarsky-apple, 2023-02-21:
> Does Matter work on Apple Homopod mini or Apple TV if the device does not have BLE, only Wi-Fi?

If the device is somehow added to the Wi-Fi network and has a commissioning window open, then I would expect yes.

If you're asking about SoftAP support, I don't believe that exists in the Matter SDK right now.

livebase, 2023-02-24:
@bzbarsky-apple 
I am testing on the Wi-Fi device and Apple TV, but the commissioning does not work properly.
Do I have to check?
Below is the log of the Wi-Fi device.

```
### User Call-back : Success to connect Wi-Fi ...
SVR: Initializing subscription resumption storage...
SVR: Server initializing...
FP: Initializing FabricTable from persistent storage
TS: Last Known Good Time: [unknown]
TS: Setting Last Known Good Time to firmware build time 2023-02-23T12:37:58
DMG: AccessControl: initializing
DMG: Examples::AccessControlDelegate::Init
DMG: AccessControl: setting
DMG: DefaultAclStorage: initializing
DMG: DefaultAclStorage: 0 entries loaded
IN: UDP::Init bind&listen port=5540
IN: UDP::Init bound to port=5540
IN: TransportMgr initialized
ZCL: Using ZAP configuration...
DMG: Failed to read stored attribute (0, 0x0000_0028, 0x0000_0005: a0
DMG: Failed to read stored attribute (0, 0x0000_0028, 0x0000_0010: a0
DMG: Failed to read stored attribute (0, 0x0000_002B, 0x0000_0000: a0
DMG: Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: a0
DMG: Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: a0
DMG: Failed to read stored attribute (1, 0x0000_0006, 0x0000_0000: a0
DMG: Failed to read stored attribute (1, 0x0000_0006, 0x0000_4003: a0
DMG: Failed to read stored attribute (1, 0x0000_0008, 0x0000_0000: a0
DMG: Failed to read stored attribute (1, 0x0000_0008, 0x0000_4000: a0
DMG: AccessControlCluster: initializing
ZCL: Initiating Admin Commissioning cluster.
DMG: Endpoint 0, Cluster 0x0000_0030 update version to d2d6a8cb
ZCL: Cluster callback: 0x0000_0030
ZCL: GeneralDiagnostics: OnDeviceReboot
DMG: Endpoint 0, Cluster 0x0000_0033 update version to 45708682
EVL: LogEvent event number: 0x0000000000000000 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0033 event id: 0x3 Epoch timestamp: 0x000000DC6ACFACC8
DMG: Endpoint 0, Cluster 0x0000_0004 update version to 581b5997
ZCL: Cluster callback: 0x0000_0004
DMG: Endpoint 0, Cluster 0x0000_0004 update version to 581b5998
ZCL: Cluster callback: 0x0000_0004
ZCL: DeviceInfoProvider is not registered
ZCL: Trying to write invalid Calendar Type
ZCL: Failed to write calendar type with error: 0x87
DMG: Endpoint 1, Cluster 0x0000_0003 update version to a73dc0f2
ZCL: Cluster callback: 0x0000_0003
ZCL: Identify attribute ID: 0x0000_0001 Type: 48 Value: 2, length 1
DMG: Endpoint 1, Cluster 0x0000_0004 update version to 42b8e4fa
ZCL: Cluster callback: 0x0000_0004
DMG: Endpoint 1, Cluster 0x0000_0004 update version to 42b8e4fb
ZCL: Cluster callback: 0x0000_0004
ZCL: Endpoint 1 On/off already set to new value
DMG: Endpoint 1, Cluster 0x0000_0008 update version to 1f95d766
ZCL: Cluster callback: 0x0000_0008
ZCL: Level Control attribute ID: 0x0000_0000 Type: 32 Value: 1, length 1
DMG: Endpoint 1, Cluster 0x0000_0300 update version to 9d53cf31
ZCL: Cluster callback: 0x0000_0300
ZCL: Color Control attribute ID: 0x0000_0007 Type: 33 Value: 0, length 2
DMG: Endpoint 1, Cluster 0x0000_0300 update version to 9d53cf32
ZCL: Cluster callback: 0x0000_0300
ZCL: Color Control attribute ID: 0x0000_0008 Type: 48 Value: 2, length 1
DMG: Endpoint 1, Cluster 0x0000_0300 update version to 9d53cf33
ZCL: Cluster callback: 0x0000_0300
ZCL: Color Control attribute ID: 0x0000_4001 Type: 48 Value: 2, length 1
DMG: Endpoint 1, Cluster 0x0000_0406 update version to 425a92fa
ZCL: Cluster callback: 0x0000_0406
DMG: Endpoint 1, Cluster 0x0000_0406 update version to 425a92fb
ZCL: Cluster callback: 0x0000_0406
IN: SecureSession[0xa9eb0]: Allocated Type:1 LSID:14572
SC: Assigned local session key ID 14572
[... 258 lines cut ...]
DMG:                    CommandPathIB =
DMG:                    {
DMG:                            EndpointId = 0x0,
DMG:                            ClusterId = 0x31,
DMG:                            CommandId = 0x0,
DMG:                    },
DMG:
DMG:                    CommandFields =
DMG:                    {
DMG:                    },
DMG:            },
DMG:
DMG:    ],
DMG:
DMG:    InteractionModelRevision = 1
DMG: },
DMG: AccessControl: checking f=0 a=p s=0xFFFFFFFB00000000 t= c=0x0000_0031 e=0 p=a
DMG: AccessControl: implicit admin (PASE)
DMG: Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0000
EM: <<< [E:52034r S:14572 M:153210354 (Ack:195525718)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
IN: (S) Sending msg 153210354 on secure session with LSID: 14572
EM: Flushed pending ack for MessageCounter:195525718 on exchange 52034r
DL: Start Scan WiFi Networks
DMG: Decreasing reference count for CommandHandler, remaining 1
```

bzbarsky-apple, 2023-02-24:
> DL: Start Scan WiFi Networks

Ok, and what happens after that?  Does the device respond to the scan?

livebase, 2023-02-24:
I don't understand why the "Start Scan WiFi Networks" command comes when the device and Apple TV are already connected to the same Wi-Fi network.

bzbarsky-apple, 2023-02-24:
Because the Apple TV does not check for that right now....

livebase, 2023-02-24:
When testing with a Linux simulated device and Apple TV, there is no log indicating that "Start Scan WiFi Networks" has been received.
What's the difference?

bzbarsky-apple, 2023-02-24:
Does the Linux simulated device have a Network Commissioning cluster with the WI feature bit set?

livebase, 2023-02-26:
I can't find anywhere in the source code to set the Network Commissioning Cluster's WI feature bit.
Excuse me, could you please tell me?

bzbarsky-apple, 2023-02-27:
> but it is displayed as no response

No idea why that's happening, if the subscription is set up correctly...  Would need a sysdiagnose from the iPad to say anything intelligent at that point.

woody-apple, 2023-06-23:
Closing this, as this has been stale for a while. 

That said, for Home issues, please file a feedback assistant ticket on Home as follows:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2023-06-23 by woody-apple