project-chip/connectedhomeip #25211: [Feature] ａｎｄｒｏｉｄ　ＮＯＣ　ｇｅｎｅｒａｔｅ　ｍｅｔｈｏｄ

luobosu, 2023-02-21:
### Feature description

Ｉ　ｗａｎｔ　ａ　ｍｅｔｈｏｄ　ｗｈｉｃｈ　Ｉ　ｃａｎ　ｕｓｅ　ｉｔ　ｔｏ　ｇｅｔ　ａ　ＮＯＣ　ｓｉｇｅｅｄ　ｂｙ　ＩＣＡＣ　ｏｎ　ａｎｄｒｏｉｄ（ｊａｖａ／ｋｏｔｌｉｎ）．
Ａｎｄ　Ｉ　ｓｅｅ　ｉＯＳ　ｓｕｐｐｏｒｔ　ｔｈｉｓ　ｍｅｔｈｏｄ　，　ｃａｎ　ａｎｙｏｎｅ　ｈｅｌｐ　ｉ　ｔ？

ｉＯＳ：
ＣＨＩＰ＿ＥＲＲＯＲ　ＭＴＲＯｐｅｒａｔｉｏｎａｌＣｒｅｄｅｎｉｔａｌｓＤｅｌｅｇａｔｅ：：ＧｅｎｅｒａｔｉｏｎａｌＣｅｒｔｉｆｉｃａｔｅ（）

### Platform

android

joonhaengHeo, 2023-02-22:
https://github.com/project-chip/connectedhomeip/blob/30188f5cf3e35a0f94265ed8c8e1f9e31202e079/src/controller/java/AndroidOperationalCredentialsIssuer.cpp#L125

https://github.com/project-chip/connectedhomeip/blob/30188f5cf3e35a0f94265ed8c8e1f9e31202e079/src/controller/java/AndroidOperationalCredentialsIssuer.cpp#L374

I think you can check this part.

luobosu, 2023-02-22:
> https://github.com/project-chip/connectedhomeip/blob/30188f5cf3e35a0f94265ed8c8e1f9e31202e079/src/controller/java/AndroidOperationalCredentialsIssuer.cpp#L125 [...]

But I do not want to use [LocalGenerateNOCChain] flow . I just want to go to the [CallbackGenerateNOCChain] flow.
                                                                 **Callback::Callback<OnNOCChainGeneration> * onCompletion)
{
    if (mUseJavaCallbackForNOCRequest)
    { CallbackGenerateNOCChain}
    else
    {LocalGenerateNOCChain
    }
}**

Becase I have generate rcac and icac by myself，then I want to use the keypair of icac to sign the NOC on the java layer , just like this:

            var issuer: ChipDeviceController.NOCChainIssuer = object :ChipDeviceController.NOCChainIssuer() {
                **override fun onNOCChainGenerationNeeded(csrInfo: CSRInfo?, attestationInfo: AttestationInfo?) {
                    //1、get the matter-iot-public-key from CSRInfo

                    //2、use a method to sign a NOC  with matter-iot-public-key and ICAC keypair  (save in nocbyte).   But I can not find the method .

                    val responseConfig : OperationalKeyConfig =
                        OperationalKeyConfig(null,racbytes,icacbytes, 
 nocbytes    ,ipkValuebytes)

                    DeviceController.onNOCChainGeneration(responseConfig,appNodeId, deviceInfo!!.vendorId)**

                }
 }

Did I use the onNOCChainGenerationNeeded by a wrong way ?

@joonhaengHeo

joonhaengHeo, 2023-02-23:
@luobosu 
The code you told me is about issuing a Controllee certificate.

https://github.com/project-chip/connectedhomeip/blob/7e728bc1bd0275f1dff8d75b2214c745047f6e01/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L250
You can add iCA from the above API's Parameters.

https://github.com/project-chip/connectedhomeip/blob/7e728bc1bd0275f1dff8d75b2214c745047f6e01/src/controller/java/src/chip/devicecontroller/ControllerParams.java#L37

And the controller will also have to set up the same root CA and iCA.
https://github.com/project-chip/connectedhomeip/blob/7e728bc1bd0275f1dff8d75b2214c745047f6e01/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L52

luobosu, 2023-02-23:
> var issuer: ChipDeviceController.NOCChainIssuer = object :ChipDeviceController.NOCChainIssuer() {
>             **override fun onNOCChainGenerationNeeded(csrInfo: CSRInfo?, attestationInfo: [...]

yes , I do this right now . But I mean the the max question is that I can not get public key from CSRINFO on java (see the !!!!!!!!!!1) , and I can not generate noc by ICAC on java layer too ( see the !!!!!!!!!!!2).

```
        var issuer: ChipDeviceController.NOCChainIssuer = object :ChipDeviceController.NOCChainIssuer() {
            **override fun onNOCChainGenerationNeeded(csrInfo: CSRInfo?, attestationInfo: AttestationInfo?) {
                //!!!!!!!!!!1、get the matter-iot-public-key from CSRInfo

                //!!!!!!!!!!!2、use a method to sign a NOC  with matter-iot-public-key and ICAC keypair  (save in nocbyte).   But I can not find the method .

                val responseConfig : OperationalKeyConfig =
                    OperationalKeyConfig(null,racbytes,icacbytes,nocbytes    ,ipkValuebytes)

                DeviceController.onNOCChainGeneration(responseConfig,appNodeId, deviceInfo!!.vendorId)**

            }
```

I modify the source code now  . let the onNOCChainGenerationNeeded method provide the nodePublicKeyFromDevice  on native code:

onNOCChainGenerationNeeded(csrInfo: CSRInfo?, attestationInfo: AttestationInfo? ,   nodePublicKeyFromDevice ByteArray? )

And I add a JNI method to generate NOC by native code .          And I solve it now . But it is better that  official offer these methods on java layer ~
@joonhaengHeo

closed 2023-03-08 by luobosu