project-chip/connectedhomeip #25237: TC-ACT-1.1] - Global mandatory attribute Eventlist  missed to list the mandatory  ActionFailed Event

Rajashreekalmane, 2023-02-22:
Problem
Based on Spec, Global mandatory attribute Eventlist  missed to list the mandatory  ActionFailed Event.

[image]

**Step to Reproduce** 
./chip-tool actions read event-list 1 1

**expected behavior** :- DUT should responds the Eventlist as 2 entries
**actual behavior** :-  DUT should responds the Eventlist as 1 entry

**system configuration**
Chip-tool and DUT - Linux
Platform : Onnetwork
Commit-ID :  e64736c9c151b45653581f749d585bddd9da1ff1 
Dut application: bridge-app

**Spec Reference**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/bridge-clusters.adoc#ref_ActionsCluster

**PFA log below**
[ACT-1.1.txt](https://github.com/project-chip/connectedhomeip/files/10800829/ACT-1.1.txt)

emargolis, 2023-02-22:
@Rajashreekalmane this issue is related to the spec issue that I opened last week:
https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6286

`ActionFailed` event currently not implemented in the SDK.

bzbarsky-apple, 2023-07-06:
Duplicate of #26721 (which has more info)

closed 2023-07-06 by bzbarsky-apple