project-chip/connectedhomeip #25299: [Android] ECM not work

nicelyjust, 2023-02-24:
### Reproduction steps

1.  Matter device commissioned by the Chip-Tool App 1
2. open commission window by ECM(Chip-Tool App 1),the UI shows manualcode & QR code
3.  failed to commissioning by the Chip-Tool App 2( or other commissioner)

if you  open commission window by BCM ,that is OK

### Platform

android

### Platform Version(s)

3951efb5cadace35e3cbd041e132eef480af1804

### Type

Manually tested with SDK

joonhaengHeo, 2023-02-24:
@nicelyjust Please upload issue log.

Do you see the error log similar to the log below?
Run command failure: ../../src/setup_payload/ManualSetupPayloadParser.cpp:49: CHIP Error 0x00000013: Integrity check failed

Linux Chiptool appears to have a similar problem, and if the Short Discriminator and Long Discriminator are the same, they will work; otherwise, they will not work.
(ex, F00 is OK, F01 is NOK.)

joonhaengHeo, 2023-02-27:
@nicelyjust 
After many attempts, it appears to appear only in certain situations. Logs are required when reproduced.

joonhaengHeo, 2023-02-27:
The comment I made last Friday seems to be wrong on the test. (Currently I'm not being reproduced.)

nicelyjust, 2023-03-06:
Thank you for reply @joonhaengHeo Currently not reproduced,But i have another question: when open BCM ,Why do I still get the previous manual pairing code ? It should be correct to generate a new value, no?

nicelyjust, 2023-03-06:
### logs

1. Commissioner-1-Android(chip-tool) log
[commissioner-Android-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/10897633/commissioner-Android-chip-tool.txt)
2. Commissioner-2-Linux(
[commissioner-linux-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/10897642/commissioner-linux-chip-tool.txt)
chip-tool) log

nicelyjust, 2023-03-07:
Thank you for your reply, Currently not reproduced too

closed 2023-03-07 by nicelyjust

joonhaengHeo, 2023-03-07:
@nicelyjust 
I think you deleted the comment.
Regarding the question in the previous comment, we created a PR that was randomly modified to create pinCode.

https://github.com/project-chip/connectedhomeip/pull/25511

nicelyjust, 2023-03-07:
yes，I removed the comment about not being a issue ，thank you for your PR @joonhaengHeo