project-chip/connectedhomeip #25302: [Workflow] TEST Test_TC_LVL_3_1 fails for LT feature

stlevkov-bosch, 2023-02-24:
### Workflow issues

Hi,

In this test [Test_TC_LVL_2_2](https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_LVL_2_2.yaml), the OnLevel attribute is set to 5. 
The next test TEST_TC_LVL_2_3 has all of the steps disabled.
The next test TEST_TC_LVL_3_1 fails due to OnLevel attribute non-null value.

Error:
```
[1677227331.633635][38358:38360] CHIP:TOO:  ***** Test Step 27 : TH sends Off command to DUT
[1677227331.636106][38358:38360] CHIP:TOO:  ***** Test Step 28 : TH sends a MoveToLevel command to the DUT with 120, transitionTime: 0, optionsMask: 0, optionsOverride: 0
[1677227331.638517][38358:38360] CHIP:TOO:  ***** Test Step 29 : TH reads CurrentLevel attribute from DUT
[1677227331.640606][38358:38360] CHIP:TOO:  ***** Test Failure: currentLevel.Value() value mismatch: expected 100 but got 1
```
It seems like the reason for that is the OnLevel attribute still holds the 5, and by the spec for the ON/OFF dependency to the LevelControl cluster:

```
Table 17. Actions on Receipt for On/Off Commands, when Associated with Level Control:
Off:
1. Temporarily store CurrentLevel.  (100)
2. Change CurrentLevel to the minimum level allowed for the device over the time period OnOffTransitionTime. (1)
3. If OnLevel is not defined, set the CurrentLevel to the stored level. (expecting 100, but got 1, due to OnLevel defined with 5)
```
After receiving the next MoveToLevel commands, the test expects the command to be discarded, due to the options bitmap ExecuteIfOf: 0, but the Level attribute has changed due to the previous OFF command to the On/Off cluster and the next test assert fails getting 1, instead of 100.

On what platform the issue was observed: _Linux GNU_
What DUT is used: _chip-all-clusters-app, bridge implementation with LT Feature_
What TH is used: _chip-tool_
What commit-sha: _[v1.0.0.2 released version](4088a77f557e8571a39338fad51a1d8eb0131d79)_
How often does the issue is reproduced: _every time_
How the issue could be resolved: _Adding write on-level null as a last step to Test_TC_LVL_2_2 fixes the issue (or a pre-condition for the LVL_3_1)_

### Platform

core (please add to version below)

### Anything else?

[LVL_3_1_Fail-chip-all-clusters.txt](https://github.com/project-chip/connectedhomeip/files/10822656/LVL_3_1_Fail-chip-all-clusters.txt)

bzbarsky-apple, 2023-02-24:
@cjandhyala

closed 2023-02-27 by cjandhyala

stlevkov-bosch, 2023-02-28:
Hi, @cjandhyala,
Unfortunately, I don't have access to this repo, or at least its giving me 404. 
Could you please check?

stlevkov-bosch, 2023-03-02:
@manjunath-grl, thanks for the reference. I can see in this [commit](https://github.com/project-chip/connectedhomeip/pull/25397/commits/7eb9a708c631eb0951959ececc5bf2fd98f82ba8#diff-cce93fe13c01c4f3bd229ba2074e2430aa88a69d7cccd42fe14bd395737da1be), that this is the expected fix.
@cjandhyala, I've got my access, but the same issue has been discussed here: [2176](https://github.com/CHIP-Specifications/chip-test-plans/issues/2176). I believe there is no need to repost it.
Thanks