project-chip/connectedhomeip #25306: Should ErrorStr be calling assertChipStackLockedByCurrentThread?

vivien-apple, 2023-02-24:
#### Problem

While working on #25155 I end up with some Tsan data races when logging errors.

It seems that https://github.com/project-chip/connectedhomeip/blob/61cf6f2368a4fc20e96d0ba87bda9a78a6028717/src/lib/support/ErrorStr.cpp#L57 use a static buffer under the hood which is not thread safe.

We could probably add `assertChipStackLockedByCurrentThread` to `ErrorStr` but I suspect that in our current examples there are various places where this is used from the main thread.

Maybe we can just allow to pass an optional application owned buffer to `ErrorStr` too.

bzbarsky-apple, 2023-02-24:
When I try making that assert, as in https://github.com/project-chip/connectedhomeip/pull/25313, various things fail.....

bzbarsky-apple, 2025-07-29:
I think https://github.com/project-chip/connectedhomeip/pull/40271 resolved this for practical purposes.