project-chip/connectedhomeip #25329: [BUG] Installation Error - Unable to checkout 'd4854d2ff06a828161d7032efb1736d0deda40b1' in submodule path 'third_party/silabs/gecko_sdk'

samleemc, 2023-02-26:
### Reproduction steps

1. cd esp_project
2. git clone --recurse-submodules git@github.com:project-chip/connectedhomeip.git

Expected - Git clone complete without error

Actual - Error

```
Submodule path 'third_party/pigweed/repo': checked out '71ba37c8a6cc65c8084078fb000e8f470cff8597'
Submodule path 'third_party/pybind11/repo': checked out '8de7772cc72daca8e947b79b83fea46214931604'
Submodule path 'third_party/qpg_sdk/repo': checked out 'eb198408bc8581702fc0fccd2e05345a1491b558'
git-lfs filter-process: git-lfs: command not found
fatal: the remote end hung up unexpectedly
fatal: Unable to checkout 'd4854d2ff06a828161d7032efb1736d0deda40b1' in submodule path 'third_party/silabs/gecko_sdk'
```

### Bug prevalence

Everytime when I tried to clone the repo

### GitHub hash of the SDK that was being used

This repo

### Platform

other

### Platform Version(s)

Mac M1 Pro 13.2.1

### Anything else?

Tried to run

`git submodule sync --recursive     `  

but still give the same error

Tried to run

`source scripts/bootstrap.sh`

but give this error

```
Checking the environment:

20230226 17:46:24 WRN Current uname (22.3.0 Darwin Kernel Version 22.3.0: Mon Jan 30 20:38:37 PST 2023; root:xnu-8792.81.3~2/RELEASE_ARM64_T6000 arm64) does not match Bootstrap uname (22.3.0 Darwin Kernel Version 22.3.0: Mon Jan 30 20:38:37 PST 2023; root:xnu-8792.81.3~2/RELEASE_ARM64_T6000 arm64), you may need to rerun bootstrap on this system
20230226 17:46:24 INF Environment passes all checks!
```

bzbarsky-apple, 2023-02-26:
@jmartinez-silabs

jmartinez-silabs, 2023-02-26:
Apparently you do not have git lfs installed with your git distribution which I find odd.

Doing git lfs install should get you pass this issue.

Also you should not have to do a recursive checkout of all the submodule.

Anyhow, I can look into why you have the error as you should not need lfs.
Gecko_sdk does have lfs support for binaries storage but those aren't used for matter and lfs should not pull any of of those without a request for it. I haven't seen before the submodule checkout trying a lfs command

@mkardous-silabs I believe you looked and validated this when we first ported to using the new gecko_sdk repo

samleemc, 2023-02-26:
@bzbarsky-apple I have fixed most of the issue myself, but when run the ninja check command, I am still stuck at `stamp obj/src/transport/tests/tests_run.stamp` . Is there anywhere I can get the log for you guys to check where it is being stuck?

bzbarsky-apple, 2023-02-27:
@samleemc If you're on an M1 Mac (which I expect you are given the uname listed above), I bet you're stuck on TestDnssd; that's sort of broken on M1 because there are network interfaces it does not expect.

Try `--args="chip_enable_dnssd_tests=false"` when you `gn gen` or equivalent?