project-chip/connectedhomeip #25332: [Android] [Door Lock]How to modify CHIPTool Android app?

nxtm555, 2023-02-27:
I try to implement Door Lock function based on CHIPTool but I don't find Door Lock arguments information.

Please let me know refer documents or source code.
I've already confirmed ChipClusters class.

**I'd like to know how to implement(what value)  Var1, 2 and 3.**

ChipClusters.class
```
        public void lockDoor(ChipClusters.DefaultClusterCallback var1, Optional<byte[]> var2, int var3) {
            this.lockDoor(this.chipClusterPtr, var1, var2, var3);
        }

        public void unlockDoor(ChipClusters.DefaultClusterCallback var1, Optional<byte[]> var2, int var3) {
            this.unlockDoor(this.chipClusterPtr, var1, var2, var3);
        }
```

***
p.s.
I refer to Lighting function which is already implemented CHIPTool.

OnOffClientFragment.kt
```
  private suspend fun sendOnCommandClick() {
    getOnOffClusterForDevice().on(object : ChipClusters.DefaultClusterCallback {
      override fun onSuccess() {
        showMessage("ON command success")
      }

      override fun onError(ex: Exception) {
        showMessage("ON command failure $ex")
        Log.e(TAG, "ON command failure", ex)
      }
    })
  }

  private suspend fun sendOffCommandClick() {
    getOnOffClusterForDevice().off(object : ChipClusters.DefaultClusterCallback {
      override fun onSuccess() {
        showMessage("OFF command success")
      }

      override fun onError(ex: Exception) {
        showMessage("OFF command failure $ex")
        Log.e(TAG, "OFF command failure", ex)
      }
    })
  }
```

ChipClusters.class
```
    public static class OnOffCluster extends ChipClusters.BaseChipCluster {
        public static final long CLUSTER_ID = 6L;

        public void off(ChipClusters.DefaultClusterCallback var1) {
            this.off(this.chipClusterPtr, var1, (Integer)null);
        }
        public void on(ChipClusters.DefaultClusterCallback var1) {
            this.on(this.chipClusterPtr, var1, (Integer)null);
        }
```

joonhaengHeo, 2023-02-27:
@nxtm555 
Please refer below codes.
https://github.com/project-chip/connectedhomeip/blob/3b80154ce7cde02bb19825da5d5b5026f90db9bd/src/controller/java/zap-generated/chip/devicecontroller/ChipClusters.java#L12485

https://github.com/project-chip/connectedhomeip/blob/3b80154ce7cde02bb19825da5d5b5026f90db9bd/src/controller/java/zap-generated/chip/devicecontroller/ChipClusters.java#L12490

yunhanw-google, 2023-03-21:
Please consider to use the API from https://github.com/project-chip/connectedhomeip/pull/25587
to implement this doorlock commands.

nxtm555, 2023-04-03:
@joonhaengHeo 
Thank you so much for your advice.
I already confirmed door lock class. I used Infineon PSoC device. It's correct operation.

I'll continue other functions implementation.

nxtm555, 2023-04-03:
@yunhanw-google 
Thank you for your kind reply.

I used wildcard and confirm correct operations.

[image]

closed 2023-04-03 by nxtm555