project-chip/connectedhomeip #25339: [XML] Update Thread Network Diagnostics Cluster XML to match updated spec

ReneJosefsen, 2023-02-27:
The data types names has been updated in Thread Network Diagnostics Cluster and must be updated in the XML as well.

**Updated**
NetworkFault -> NetworkFaultEnum

**Added**
- RoutingRoleEnum
- NeighborTableStruct
- RouteTableStruct
- SecurityPolicyStruct
- OperationalDatasetComponentsStruct

Spec PR: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6169 (already merged)

bzbarsky-apple, 2023-05-26:
> SecurityPolicyStruct

This is actually called SecurityPolicy in the spec, after https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6338

> OperationalDatasetComponentsStruct

Likewise, it's OperationalDatasetComponents

ReneJosefsen, 2023-05-26:
Yeah, I forgot to update these issues after we got the revert merged. Good catch, my mistake 😅

closed 2023-05-30 by andy31415