project-chip/connectedhomeip #25360: Matter Accessory(device) not responding after commissioning with Apple Mini Homepad

gksdlfxor, 2023-02-28:
### Reproduction steps

1. The accessory(device) is using lighting-app in FreeRTOS and it is working well with Android chip-tool or Linux chip-tool.
2. There are IPad(controller), Apple Homepad(Hub) and the accessory(device).
3. The commissioning of accessory(device) is working well  with Apple Mini Homepad.
4. But Ipad's Home show "Matter Accessory not responding".
5. At that time, it's log always show "**DMG: Error retrieving data from clusterId: 0x0000_001D, err = b**"
6. Detail log is the follows.

DMG: Received command for Endpoint=0 Cluster=0x0000_003E Command=0x0000_0009
[IMClusterCommandHandler.cpp:DispatchSingleClusterCommand:1132] ----
[IMClusterCommandHandler.cpp:DispatchServerCommand:928] ---- aCommandPath.mCommandId : 0x9
ZCL: OpCreds: Received an UpdateFabricLabel command
FP: Metadata for Fabric 0x1 persisted to storage.
DMG: Endpoint 0, Cluster 0x0000_003E update version to 9b2c6951
DMG: The dirty path intersects the cluster we are currently reporting; reset the iterator to the beginning of that cluster
DMG: Cannot merge the new path into any existing path, create one.
DMG: Command handler moving to [ Preparing]
DMG: Command handler moving to [AddingComm]
DMG: Command handler moving to [AddedComma]
DMG: Decreasing reference count for CommandHandler, remaining 0
EM: <<< [E:38647r M:186498174 (Ack:134116388)] (S) Msg TX to 1:53E0A2037DA7608B [3868] --- Type 0001:09 (IM:InvokeCommandResponse)
IN: (S) Sending msg 186498174 on secure session with LSID: 31375
[UDPEndPointImplLwIP.cpp:SendMsgImpl:242] da16x_sendto_xx: pbuf=0x000a0048, netif=0x000cc9d0, destPort=55218
DMG: Command handler moving to [CommandSen]
DMG: Command handler moving to [AwaitingDe]
DL: Long dispatch time: 130 ms, for event type 3
[UDPEndPointImplLwIP.cpp:SendMsgImpl:242] da16x_sendto_xx: pbuf=0x000a06d8, netif=0x000cc9d0, destPort=5353
DMG: Building Reports for ReadHandler with LastReportGeneration = lu DirtyGeneration = lu
DMG: <RE:Run> Cluster 4, Attribute 0 is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_0000 (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fffc is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFFC (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fffd is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFFD (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fff8 is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFF8 (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fff9 is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFF9 (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fffb is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFFB (expanded=1)
DMG: <RE:Run> Cluster 1d, Attribute 0 is dirty
DMG: Reading attribute: Cluster=0x0000_001D Endpoint=0 AttributeId=0x0000_0000 (expanded=1)
DMG: <RE:Run> Cluster 1d, Attribute 1 is dirty
DMG: Reading attribute: Cluster=0x0000_001D Endpoint=0 AttributeId=0x0000_0001 (expanded=1)
DMG: Error retrieving data from clusterId: 0x0000_001D, err = b

DMG: <RE:Run> We cannot put more chunks into this report. Enable chunking.
DMG: <RE:Run> first cluster event is too big so that it fails to fit in the packet!
DMG: Fetched 0 events
DMG: <RE> Sending report (payload has 984 bytes)...
EM: <<< [E:38645r M:186498175 (Ack:134116386)] (S) Msg TX to 1:53E0A2037DA7608B [3868] --- Type 0001:05 (IM:ReportData)
IN: (S) Sending msg 186498175 on secure session with LSID: 31375
[UDPEndPointImplLwIP.cpp:SendMsgImpl:242] da16x_sendto_xx: pbuf=0x0009f9b8, netif=0x000cc9d0, destPort=55218
DMG: IM RH moving to [AwaitingReportResponse]
DMG: <RE> ReportsInFlight = 1 with readHandler 0, RE has more messages
DMG: Building Reports for ReadHandler with LastReportGeneration = lu DirtyGeneration = lu
DMG: <RE:Run> Cluster 3e, Attribute 1 is dirty
DMG: Reading attribute: Cluster=0x0000_003E Endpoint=0 AttributeId=0x0000_0001 (expanded=0)
DMG: <RE> Sending report (payload has 274 bytes)...
EM: <<< [E:38646r M:186498176 (Ack:134116387)] (S) Msg TX to 1:53E0A2037DA7608B [3868] --- Type 0001:05 (IM:ReportData)
IN: (S) Sending msg 186498176 on secure session with LSID: 31375
[UDPEndPointImplLwIP.cpp:SendMsgImpl:242] da16x_sendto_xx: pbuf=0x000a1a88, netif=0x000cc9d0, destPort=55218
DMG: IM RH moving to [AwaitingReportResponse]
DMG: <RE> ReportsInFlight = 2 with readHandler 1, RE has no more messages
EM: >>> [E:38647r M:134116389 (Ack:186498174)] (S) Msg RX from 1:53E0A2037DA7608B [3868] --- Type 0000:10 (SecureChannel:StandaloneAck)
EM: Found matching exchange: 38647r, Delegate: 0
EM: Rxd Ack; Removing MessageCounter:186498174 from Retrans Table on exchange 38647r
EM: >>> [E:38648r M:134116390] (S) Msg RX from 1:53E0A2037DA7608B [3868] --- Type 0001:02 (IM:ReadRequest)
EM: Handling via exchange: 38648r, Delegate: 0xadb1c
[... 453 lines cut ...]
DMG: <RE:Run> Cluster 4, Attribute fff9 is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFF9 (expanded=1)
DMG: <RE:Run> Cluster 4, Attribute fffb is dirty
DMG: Reading attribute: Cluster=0x0000_0004 Endpoint=0 AttributeId=0x0000_FFFB (expanded=1)
DMG: <RE:Run> Cluster 1d, Attribute 0 is dirty
DMG: Reading attribute: Cluster=0x0000_001D Endpoint=0 AttributeId=0x0000_0000 (expanded=1)
DMG: <RE:Run> Cluster 1d, Attribute 1 is dirty
DMG: Reading attribute: Cluster=0x0000_001D Endpoint=0 AttributeId=0x0000_0001 (expanded=1)
DMG: Error retrieving data from clusterId: 0x0000_001D, err = b

DMG: <RE:Run> We cannot put more chunks into this report. Enable chunking.
DMG: Fetched 0 events
DMG: <RE> Sending report (payload has 984 bytes)...

### Bug prevalence

always

### GitHub hash of the SDK that was being used

6f041acf0e9afdb740be05e009fdc8fffe6cdb78

### Platform

freeRTOS

bzbarsky-apple, 2023-02-28:
> At that time, it's log always show "DMG: Error retrieving data from clusterId: 0x0000_001D, err = b"

This is CHIP_ERROR_NO_MEMORY and is expected when chunking reads/subscribes.

@gksdlfxor could you please follow instructions below to file a feedback assistant ticket with Apple?   Make sure to include the Matter device log and sysdiagnoses from the iPad and HomePod.  Please comment here with the ticket ID and I will follow up on it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

woody-apple, 2023-06-23:
Closing, as this should be followed up with via feedback assistant.

closed 2023-06-23 by woody-apple