project-chip/connectedhomeip #25365: [BUG] iOS 16.3.1 operate abnormally

Throm-Ka, 2023-02-28:
### Reproduction steps

iphone : 16.3.1
homepod mini : 16.3.2

Pairing device unsuccessful.  Apple Home seems crash ?

When I use another iphone : 16.2 , good running.

### Bug prevalence

Whenever

### GitHub hash of the SDK that was being used

tag: 1.0.0.2

### Platform

freeRTOS

bzbarsky-apple, 2023-02-28:
@Throm-Ka Can you please follow instructions below to file a feedback assistant ticket? Then follow up here with the ticket id, please.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

Throm-Ka, 2023-03-01:
iPhone update to 16.4 , there's no problem of pairing

closed 2023-03-22 by Throm-Ka