project-chip/connectedhomeip #25367: While testing Matter Ecosystem with Google Nest Hub and Home pod Mini observed below issues

shrtui, 2023-02-28:
### Reproduction steps

While testing with Apple and Google Nest Hub with Light application seen few observations as below:

Scenario -1

1.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
2.	Manually Turn OFF the light - not through Siri 
3.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here it is moving to 100%

Scenario -2

1.	Do Multi fabric – connect Apple and Google Nest
2.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
3.	Manually Turn OFF the light - not through Siri 
4.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here Apple is moving to 100%
and Google Nest light intensity is 1%

Scenario -3

1.	With Apple – Say change color to Black – Light is turning OFF
2.	With Apple – Say change color to White – Light is turning ON
3.	With Google Nest - Say change color to Black – Light is turning OFF
4.	With Google Nest - Say change color to White – Light is not turning ON
5.	With Apple – When light is ON Say Change light intensity light is turning OFF and vice versa also happening.

While testing with Apple with Thermostat application seeing issue with Auto mode (Heat/Cool)
1.	There are four options while testing Thermostat Application Heat, Cool, Auto and OFF
2.	If Auto mode is selected APP is crashing when trying open again APP is no opening
3.	Even with Multi Admin same issue observed , if in Google Nest Auto mode is selected in Apple Thermostat APP is getting closed
While testing with Apple and Google Nest Hub with Light application seen few observations as below:

Note: ** Consolidating yesterday sent issues also in this mail itself.

Scenario -1

1.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
2.	Manually Turn OFF the light - not through Siri 
3.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here it is moving to 100%

Scenario -2

1.	Do Multi fabric – connect Apple and Google Nest
2.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
3.	Manually Turn OFF the light - not through Siri 
4.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here Apple is moving to 100%
and Google Nest light intensity is 1%

Scenario -3

1.	With Apple – Say change color to Black – Light is turning OFF
2.	With Apple – Say change color to White – Light is turning ON
3.	With Google Nest - Say change color to Black – Light is turning OFF
4.	With Google Nest - Say change color to White – Light is not turning ON
5.	With Apple – When light is ON Say Change light intensity light is turning OFF and vice versa also happening.

While testing with Apple with Thermostat application seeing issue with Auto mode (Heat/Cool)
1.	There are four options while testing Thermostat Application Heat, Cool, Auto and OFF
2.	If Auto mode is selected APP is crashing when trying open again APP is no opening
3.	Even with Multi Admin same issue observed , if in Google Nest Auto mode is selected in Apple Thermostat APP is getting closed
4.	While testing Multi admin with Apple and Google Nest “*ERR*EN-RSI:Send fail” issue observed

While testing with Apple  with Windows application seeing few issues please clarify

1.	May I know what is the actual command should I use while testing with Siri, like “Open blinds to 10% or Close blinds to 80%” Or only Open term should I use for window application because 
2.	When the blinds are completely Open , If I say “Open blind to 40% - blinds are open to 40%” and If I say “Close blind to 80%- blind is setting it to Open 80%”
3.	Also between each commands(through Siri) is there any time gap we should maintain why because if I say commands one after another it is not responding properly
4.	In between execution “*ERR*EN-RSI:Send fail” issue observed and application is not responding

While testing with Apple and Google Nest Hub with Light application seen few observations as below:

Note: ** Consolidating yesterday sent issues also in this mail itself.

Scenario -1

1.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
2.	Manually Turn OFF the light - not through Siri 
3.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here it is moving to 100%

Scenario -2

1.	Do Multi fabric – connect Apple and Google Nest
2.	Turn  ON the light - do few changes in color/light intensity like change light color to red or change light intensity to 70%
3.	Manually Turn OFF the light - not through Siri 
4.	Turn ON the light through Siri – Ideally light intensity should be previous intensity which is 70% but here Apple is moving to 100%
and Google Nest light intensity is 1%

Scenario -3

1.	With Apple – Say change color to Black – Light is turning OFF
2.	With Apple – Say change color to White – Light is turning ON
3.	With Google Nest - Say change color to Black – Light is turning OFF
4.	With Google Nest - Say change color to White – Light is not turning ON
5.	With Apple – When light is ON Say Change light intensity light is turning OFF and vice versa also happening.

While testing with Apple with Thermostat application seeing issue with Auto mode (Heat/Cool)
1.	There are four options while testing Thermostat Application Heat, Cool, Auto and OFF
2.	If Auto mode is selected APP is crashing when trying open again APP is no opening
3.	Even with Multi Admin same issue observed , if in Google Nest Auto mode is selected in Apple Thermostat APP is getting closed
4.	While testing Multi admin with Apple and Google Nest “*ERR*EN-RSI:Send fail” issue observed

1.	May I know what is the actual command should I use while testing with Siri, like “Open blinds to 10% or Close blinds to 80%” Or only Open term should I use for window application because 
2.	When the blinds are completely Open , If I say “Open blind to 40% - blinds are open to 40%” and If I say “Close blind to 80%- blind is setting it to Open 80%”
3.	Also between each commands(through Siri) is there any time gap we should maintain why because if I say commands one after another it is not responding properly
4.	In between execution “*ERR*EN-RSI:Send fail” issue observed and application is not responding

1.	May I know what is the actual command should I use while testing with Siri, like “Open blinds to 10% or Close blinds to 80%” Or only Open term should I use for window application because 
2.	When the blinds are completely Open , If I say “Open blind to 40% - blinds are open to 40%” and If I say “Close blind to 80%- blind is setting it to Open 80%”
3.	Also between each commands(through Siri) is there any time gap we should maintain why because if I say commands one after another it is not responding properly

### Bug prevalence

100

### GitHub hash of the SDK that was being used

1.0.0.2

### Platform

efr32

bzbarsky-apple, 2023-02-28:
@shrtui Can you please follow instructions below to file a feedback assistant ticket? Then follow up here with the ticket id, please.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

woody-apple, 2023-06-23:
Closing, as this Apple Home issue should be followed up on via feedback assistant.