project-chip/connectedhomeip #25382: [BUG] Chip-App fails to execute Test case TC-PRS-3.1

sholagi, 2023-02-28:
### Reproduction steps

1. Open Terminal 1 and run 'rm -rf /tmp/chip_*;./chip-app1 --trace_file "/logs/trace_log_2023-02-28_19.47.05_0xb14e1214ed9efe87_SIMULATED_APP_TEST_PAIRING.log" --trace_decode 1`
2. Open Terminal 2 and run `rm -rf /tmp/chip*;./chip-tool pairing code 305414945 MT:-24J0AFN00KA0648G00`
3. Ctrl+C on Terminal 1.
4. In Terminal 1, run `./chip-app1 --command Test_TC_PRS_3_1_Simulated --secured-device-port 5540 --trace_file "/logs/trace_log_2023-02-28_19.47.14_0xb14e1214ed9efe87_SIMULATED_APP_TEST_Test_TC_PRS_3_1_Simulated.log" --trace_decode 1`
5. In Terminal 2, run the below commands:
 ./chip-tool pressuremeasurement read measured-value 305414945 0
 ./chip-tool pressuremeasurement read min-measured-value 305414945 0
 ./chip-tool pressuremeasurement read max-measured-value 305414945 0
 ./chip-tool pressuremeasurement read tolerance 305414945 0
 ./chip-tool pressuremeasurement read scaled-value 305414945 0
 ./chip-tool pressuremeasurement read min-scaled-value 305414945 0
 ./chip-tool pressuremeasurement read max-scaled-value 305414945 0
 ./chip-tool pressuremeasurement read scaled-tolerance 305414945 0
 ./chip-tool pressuremeasurement read scale 305414945 0
5. Observe in Terminal 1 that `chip-app` seem to skip step "read scale" and move to next step to prompt user. Chip-app does not seem to respond to the request from the controller.
6. Observe in Terminal 2 that `chip-tool` fails. 

Logs of test step to read scale attribute are attached from `chip-app` and `controller`

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

b89e83be43dde7a79d90823f4bc1279eb53f76de

### Platform

raspi

sholagi, 2023-02-28:
Logs 
[TC-PRS-3.1-Chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/10854510/TC-PRS-3.1-Chip-tool.txt)
[TC-PRS-3.1-Chip-App.txt](https://github.com/project-chip/connectedhomeip/files/10854512/TC-PRS-3.1-Chip-App.txt)

raju-apple, 2023-02-28:
@krypton36 can you take a look please ?

krypton36, 2023-03-08:
I believe the issue here is that when the following step if UserPrompt, the main app thread is being blocked by the the waiting for the user to put in the response. If I quickly hit yes to all the user prompts, the response is finally sent. 
[Quick_Yes_Delayed_Report.txt](https://github.com/project-chip/connectedhomeip/files/10923006/Quick_Yes_Delayed_Report.txt)

To resolve this issue, we need a way for the UserPrompt to not block the response or have the NexTest() wait until the Report has been sent. @vivien-apple can you please take a look?

krypton36, 2023-03-08:
@vivien-apple I created a PR above, can you please check to see if that fix is ok?

PR #25576 (2023-03-20): Resolve race condition before UserPrompt in simulated tests.
- Fixes #25382  
- Updated the next test to be executed as scheduled work so it does not block the read/write/command to complete when a user prompt is executed on the following step.

Tests:
- Validated that the issue filed no longer occurs.
- Validated that all steps are completed in the test.
- Validated that user prompt is still accepted and continue after typing it in.
closed 2023-03-20 by andy31415