project-chip/connectedhomeip #25385: [BUG] source scripts/bootstrap.sh fails on Debian Bookworm

s-marios, 2023-03-01:
### Reproduction steps

On Debian Bookworm, python ver. 3.11.2, the bootstrap script fails.

To reproduce, follow the "building matter" document,

0. install prerequisites
1.  git clone --recurse-submodules 
2. source scripts/bootstrapt.sh

This esults in the log file attached. [console_output.txt](https://github.com/project-chip/connectedhomeip/files/10856825/console_output.txt)

In sort, it fails to build (or actually install?) the wheels for gevent and greenlet.

I have no idea why this happens or how I could proceed from there.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

e75d35a7581ced7152a4642e070c21f942ab66d7

### Platform

core

### Anything else?

This change in Debian about having python interpreters marked as  [externally managed](https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.en.html#python3-pep-668) may be of relevance.

Also, sometime in December I had no problem with the bootstrap script and was in fact able to build matter successfully. However, it seems that further updates in Debian Testing broke the bootstrap script.

Using a venv (pyton3.11 -m venv chip-venv) led to the same result.

Can you please look into this? Any insights are appreciated.

s-marios, 2023-03-01:
Small update, trying the procedure in a Debian Stable lxc container works as expected, so Debian Testing is definitely the suspect here.

andy31415, 2023-03-01:
× Building wheel for gevent (pyproject.toml) did not run successfully.

We seem to have had this problem occasionally due to esp32 requirement (gdbui -> gevent). @dhrishi  ... any chance we can avoid esp32 requiring gdbui?

@s-marios could you test out what happens if you edit `scripts/setup/requirements.esp32.txt` and comment out (place a `#` in from of) the `gdbui` line?

s-marios, 2023-03-02:
> @s-marios could you test out what happens if you edit `scripts/setup/requirements.esp32.txt` and comment out (place a `#` in from of) the `gdbui` line?

This fixes things and activate.sh/bootstrap.sh seem to work.

I guess this does not affect normal development and the only thing that we miss out on is being able to use gdbui, am I right?

mspang, 2023-03-02:
Do we perhaps just need to update the constraints.txt ?

mspang, 2023-03-02:
Doesn't seem like there is a newer version of gevent, but I posted a PR to update anyway since it hasn't been done in a while:

https://github.com/project-chip/connectedhomeip/pull/25441

mspang, 2023-03-02:
greenlet was updated. It's still possible this will resolve your issue. Mind trying it @s-marios  ?

andy31415, 2023-03-02:
if we don't install gdbui I believe esp32 builds may fail because it is mandatory there (builds switch to chip venv and then builds may completely fail, even if idl exports.sh would install it).

If we make changes like commenting out gdbui, we have to ensure esp32 builds work (generally CI would check that, so that may be sufficient)

andy31415, 2023-03-02:
Tried the fix in 25441 on my arch machine (which has the gdbui issue), no help unfortunately:

```
...
      src/gevent/greenlet.c: In function ‘__pyx_f_6gevent_9_greenlet_get_generic_parent’:
      src/gevent/greenlet.c:16215:26: error: ‘PyGreenlet’ {aka ‘struct _greenlet’} has no member named ‘parent’
      16215 |   __pyx_t_1 = ((__pyx_v_s->parent != NULL) != 0);
...
```

andy31415, 2023-03-02:
Just checked in our latest image (with latest esp32) and gdbui is NOT in ` /opt/espressif/esp-idf/requirements.txt`
maybe just removing is acceptable after all.

andy31415, 2023-03-02:
I was missing a g in my grep. It is there after all as `gdbgui==0.13.2.0` :( 

We could patch it in as part of our build image building, however that is probably not maintainable. I would much rather have esp-idf updated. @dhrishi

mspang, 2023-03-02:
Ok, well, it sounds like this may be blocked on resolving some upstream dependency issues.

In order to fix compatibility issues, the dependencies must make a new release. So they must be updated. But we are blocking a handful of updates (presumably due to some breakage)

requirements.esp32.txt:kconfiglib==13.7.1
requirements.esp32.txt:construct==2.10.54
requirements.esp32.txt:gdbgui==0.13.2.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'
requirements.mbed.txt:pytest==6.2.5 ; platform_machine != 'aarch64' and sys_platform == 'linux'
requirements.mbed.txt:mbed-ls==1.8.11 ; platform_machine != 'aarch64' and sys_platform == 'linux'
requirements.telink.txt:future==0.18.3
requirements.telink.txt:pypng==0.0.21
requirements.telink.txt:PyQRCode==1.2.1
requirements.txt:build==0.8.0
requirements.txt:mypy==0.971
requirements.txt:mypy-protobuf==3.2.0
requirements.txt:protobuf==3.20.1
requirements.txt:types-protobuf==3.19.22

Trying to hold these back together probably won't work, as the configuration eventually becomes unsatisfiable.

andy31415, 2023-03-02:
Just removing GDBUI will not work: 

```
# ./scripts/build/build_examples.py --target esp32-m5stack-all-clusters build
...
2023-03-02 21:30:35 INFO    Environment looks good, you are ready to go!
2023-03-02 21:30:35 INFO
2023-03-02 21:30:35 WARNING The following Python requirements are not satisfied:
2023-03-02 21:30:35 WARNING gdbgui==0.13.2.0
2023-03-02 21:30:35 WARNING To install the missing packages, please run "/opt/espressif/esp-idf/install.sh"
2023-03-02 21:30:35 WARNING Diagnostic information:
2023-03-02 21:30:35 WARNING     IDF_PYTHON_ENV_PATH: /opt/espressif/tools/python_env/idf4.4_py3.9_env
2023-03-02 21:30:35 WARNING     Python interpreter used: /home/vscode/pigweed/env/pigweed-venv/bin/python
....
```

s-marios, 2023-03-03:
> if we don't install gdbui I believe esp32 builds may fail because it is mandatory there (builds switch to chip venv and then builds may completely fail, even if idl exports.sh would install it).

That's unfortunate, as it also happens that I'm targeting esp32.

> Tried the fix in 25441 on my **arch machine** (which has the gdbui issue)

So, this is not a Debian problem, it seems to be a python 3.11 problem I suppose? If that's the case, let's change the title of this issue to something more appropriate.

commit a6bb7185 (2023-05-12): ESP32: avoid installing gdbgui when not needed (#26542)

ESP-IDF v4.4.4 requires gdbgui only when Python before 3.11 is used (see
https://github.com/espressif/esp-idf/commit/3974be7fec1ea6c529ecbee795c0152b42b61d55).
Avoid installing it when not needed.

Fixes: #25385
commit 4aef92fa (2023-08-08): [Cherrypick] CI: Fix for v1.1-branch CI, broken due to gdbgui (#28507)

* ESP32: avoid installing gdbgui when not needed (#26542)

ESP-IDF v4.4.4 requires gdbgui only when Python before 3.11 is used (see
https://github.com/espressif/esp-idf/commit/3974be7fec1ea6c529ecbee795c0152b42b61d55).
Avoid installing it when not needed.

Fixes: #25385

* Remove gdbgui requirement for esp32 (#28007)

* Remove gdbgui requirement for esp32

* Fix qemu

* Fix chef as well

---------
commit eb231872 (2023-08-25): Support for Hardware Secure Module (HSM) using Infineon Optiga Trust M  (#28397)

* Added support for Optiga Trust M.

* * Added no warning flag when applying patch for optiga-trust-m.
* Add the optiga_lib_config_mtb.h

* 1)Updated README.md for psoc6 lock-app example

2)Added infineon_trustm_provisioning.md

* 1)Updated README.md for psoc6 lock-app

* 1)Updated README.md for psoc6 lock-app

* 1)Updated optiga-trust-m submodule

2)Updated README.md for psoc6 lock-app

* 1)Updated DeviceAttestationCredsExampleTrustM.cpp

2)Updated the argument with infineon added

* 1)Updated CHIPCryptoPALHsm_H [... cut]
closed 2023-05-12 by andy31415