project-chip/connectedhomeip #25391: [mbed-mcu-boot] mbed-mcu-boot repo include too much submodule

shchen-Lab, 2023-03-01:
### Reproduction steps

Hi Team

mbed-mcu-boot platform include too much sub module，this ’s badly affected sync submodule.
[image]

### Platform

other

### Type

Platform validated

bzbarsky-apple, 2023-03-01:
@shchen-Lab what exact command did you run to cause all those to be cloned?  `third_party/mbed-mcu-boot/repo/boot/espressif/hal/esp-idf/` is empty for me when I run either a normal bootstrap or `scripts/checkout_submodules.py`

shchen-Lab, 2023-03-02:
git submodule update --recursive --init

bzbarsky-apple, 2023-03-06:
Yes, "don't do that".  In particular, that `--recursive` bit: don't do it.

closed 2023-03-14 by shchen-Lab