project-chip/connectedhomeip #25424: [BUG] How to read multiple attribute in darwin platform

joonhaengHeo, 2023-03-02:
### Reproduction steps

There is no way to read multiple attributes at once from the Darwin platform.

We have confirmed that one attribute can be read using the readAttributeWithEndpointId of MTRBaseDeviceClass, and that endpointId, clusterId, attributeId can be used as null value(nil) parameter to read all attributes. However, it is impossible to deliver in List form like Android.

https://github.com/project-chip/connectedhomeip/blob/1a1d0c3c23d78f8fa72c7ef0d6ad88ff05ef0224/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L567

### Bug prevalence

always

### GitHub hash of the SDK that was being used

1a1d0c3c23d78f8fa72c7ef0d6ad88ff05ef0224

### Platform

darwin

bzbarsky-apple, 2023-03-06:
> There is no way to read multiple attributes at once from the Darwin platform.

That is correct.  That's not an API the Darwin SDK exposes at the moment.

Exposing such an API would require some thought about where the responsibility lies for dealing with "too large" reads (i.e. involving too many paths) and how those should be handled.

joonhaengHeo, 2023-03-07:
@bzbarsky-apple 
Thank you for your answer.

As far as I know, up to 9 multiple attributes can be read. (I received an answer to Slack question before.) Wouldn't it be right to support Multiple Attribute Read even if there is a limit of 9 Reads?

joonhaengHeo, 2023-03-07:
Do you have a plan to add multiple attribute reading API to darwin?

bzbarsky-apple, 2023-03-08:
Me personally, in the next few weeks, no.  But I would not object to someone else doing it, as long as the API is reasonable.

PR #25840 (2023-04-18): [Darwin] Implement multiple read and subscribe
Fixes #25424 

1. Implement multiple read attribute or event.
2. Implement multiple subscribe attribute or event.
closed 2023-04-18 by bzbarsky-apple