project-chip/connectedhomeip #25426: [BUG] CASESession timed out

lancersky, 2023-03-02:
### Reproduction steps

According to the topology, chip-tool run on ubuntu 20.04.4 LTS, OTBR(ot-br-posix: 2023-02-21 ec649a8ec2) run on Raspi 4B, the Thread device is EFR32MG24 breakout board base on GSDK4.2.0.
If both OTBR and Controller are running on the same device, it will be successful.
[image]
[pairing failed.txt](https://github.com/project-chip/connectedhomeip/files/10869932/pairing.failed.txt)
```
[1677749485.105030][3803578:3803580] CHIP:SC: CASESession timed out while waiting for a response from the peer. Current state was 1
[1677749485.105047][3803578:3803580] CHIP:IN: SecureSession[0x7f40dc032540]: Released - Type:2 LSID:38757
[1677749485.105068][3803578:3803580] CHIP:CTL: Device connection failed. Error ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:295: CHIP Error 0x00000032: Timeout
[1677749485.105079][3803578:3803580] CHIP:CTL: Error on commissioning step 'FindOperational': '../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:295: CHIP Error 0x00000032: Timeout'
[1677749485.105086][3803578:3803580] CHIP:CTL: Failed to perform commissioning step 18
[1677749485.105092][3803578:3803580] CHIP:CTL: Going from commissioning step 'FindOperational' with lastErr = '../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:295: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
[1677749485.105101][3803578:3803580] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = '../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:295: CHIP Error 0x00000032: Timeout'
[1677749485.105107][3803578:3803580] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
[1677749485.105115][3803578:3803580] CHIP:TOO: Device commissioning Failure: ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:295: CHIP Error 0x00000032: Timeout
[1677749485.105320][3803578:3803578] CHIP:CTL: Shutting down the commissioner
[1677749485.105330][3803578:3803578] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1677749485.105355][3803578:3803578] CHIP:DIS: Closing all BLE connections
[1677749485.105361][3803578:3803578] CHIP:IN: SecureSession[0x7f40dc029400]: MarkForEviction Type:1 LSID:38756
[1677749485.105366][3803578:3803578] CHIP:SC: SecureSession[0x7f40dc029400]: Moving from state 'kActive' --> 'kPendingEviction'
[1677749485.105380][3803578:3803578] CHIP:IN: SecureSession[0x7f40dc029400]: Released - Type:1 LSID:38756
[1677749485.105388][3803578:3803578] CHIP:CTL: Shutting down the controller
```

### Bug prevalence

It will fail every time according to the topology to test.

### GitHub hash of the SDK that was being used

2022-12-21-4088a77f5 - (HEAD, tag: v1.0.0.2)

### Platform

efr32, raspi

bzbarsky-apple, 2023-03-06:
@Damian-Nordic @jepenven-silabs

Damian-Nordic, 2023-03-06:
@lancersky chip-tool has succeeded to find an IP address of the device but apparently it failed to route the message correctly (or a response hasn't been sent back - it's hard to tell without the device logs). I suggest first checking with `ip -6 route` if the routing table on your PC looks correct, that is, if the Thread network prefix is routed via OTBR. If not, please read "Ping from a Linux/macOS host" section from https://openthread.io/codelabs/openthread-border-router#6 to configure accepting router advertisements on your PC.

If the routing seems fine, then a packet capture from the OTBR may be needed to further analyse this issue.

jepenven-silabs, 2023-03-06:
There might be multiples causes to this issue.... Is it on a new network? what are the logs on the device side? did the NodeId get reused causing the SRP to fail on the OTBR ? Like Damian said, this is hard to help/debug without the device logs...