project-chip/connectedhomeip #25433: Matter Test-Harness

stefano055415, 2023-03-02:
I have a problem with TH version 2.6

If I load the PICS files the correct tests are not selected

Steps to reproduce it
Modify the project by uploading the Pics file:

[image]

I press on Run Test:

[image]

I create a new test:

[image]

But the selected tests do not match those of the pics file

what am I doing wrong thanks

this is the pics file

[On-Off Cluster Test Plan.xml.zip](https://github.com/espressif/esp-matter/files/10872949/On-Off.Cluster.Test.Plan.xml.zip)

yangboHDU, 2023-03-03:
How do you get your pics file? My TH2.6 only has some test cases at present. I can't find the file when uploading the pics file

cjandhyala, 2023-03-03:
Pls file the ticket in the Matter TH repo https://github.com/CHIP-Specifications/chip-certification-tool, closing the ticket here as this is not SDK issue.

closed 2023-03-03 by cjandhyala

stefano055415, 2023-03-03:
@zmeryy 
The file is present in the csa spec https://groups.csa-iot.org/wg/members-all/document/folder/2269

@cjandhyala 
This repo is private and I don't know who to ask for access, could you help me?

cjandhyala, 2023-03-03:
@stefano055415 Could you pls reach out to CSA staff or Chris LaPré for help.