project-chip/connectedhomeip #25442: [Platform] Android failed to commission when BLE Indication not supported by Device

swan-amazon, 2023-03-02:
### Reproduction steps

The Moross plug device launch firmware was based an old copy of the SDK, provided by Realtek, which did not support enabling BLE Indications. When the command is sent to the device to enable BLE Indications, the device returns an unsupported error and the command times out. As seen below.

```
29.193620000; ATT Write Transaction (Client Characteristic Configuration: Notifications=Disabled, Indications=Enabled; Reserved (0xC0))       ; 02 00; 
29.193620000;   ATT Write Request Packet (Client Characteristic Configuration: Notifications=Disabled, Indications=Enabled)                   ; 12 11 00 02 00; 
29.193620000;     L2CAP SDU (Basic, Service=ATT)                                                                                              ; 12 11 00 02 00; 
29.193620000;       L2CAP B-Frame (Service=ATT, ACL-U Flow Events)                                                                            ; 05 00 04 00 12 11 00 02 00; 
29.181786000;         HCI Number Of Completed Packets (Connection=0x0002, Packets=1)                                                          ; 13 05 01 02 00 01 00; 
29.193620000;         HCI ACL Data OUT                                                                                                        ; 02 00 09 00 05 00 04 00 12 11 00 02 00; 
29.204326000;   ATT Error Response Packet (Reserved (0xC0))                                                                                   ; 01 12 11 00 C0; 
29.204326000;     L2CAP SDU (Basic, Service=ATT)                                                                                              ; 01 12 11 00 C0; 
29.204326000;       L2CAP B-Frame (Service=ATT)                                                                                               ; 05 00 04 00 01 12 11 00 C0; 
29.204326000;         HCI ACL Data IN                                                                                                         ; 02 20 09 00 05 00 04 00 01 12 11 00 C0; 
```

```
03-02 15:39:13.223 14051 15538 D IN      : BleConnectionComplete: endPoint 0x7303eb6780
03-02 15:39:13.223 14051 15538 D DL      : Received GetMTU
03-02 15:39:13.223 14051 15538 D DL      : Received SendWriteRequest
03-02 15:39:13.225 15251 15263 D BtGatt.GattService: writeCharacteristic() - trying to acquire permit.
03-02 15:39:13.226 14051 15538 D IN      : SecureSession[0xb40000767264bd00]: Allocated Type:1 LSID:26500
03-02 15:39:13.226 14051 15538 D SC      : Assigned local session key ID 26500
03-02 15:39:13.226 14051 15538 D EM      : <<< [E:50518i S:0 M:108446742] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
03-02 15:39:13.226 14051 15538 D IN      : (U) Sending msg 108446742 to IP address 'BLE'
03-02 15:39:13.226 14051 15538 D IN      : Message appended to BLE send queue
03-02 15:39:13.226 14051 15538 D SC      : Sent PBKDF param request
03-02 15:39:13.237 15251 15283 D BtGatt.GattService: onWriteCharacteristic() - increasing permit for address=F4:EA:79:33:AC:B5
03-02 15:39:13.237 14051 15538 D DL      : Received SubscribeCharacteristic
03-02 15:39:13.238 14051 15538 D BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true
03-02 15:39:13.252 14051 15538 E AndroidBleManager: onDescriptorWrite for 00002902-0000-1000-8000-00805f9b34fb failed with status: 192
03-02 15:39:13.252 14051 15538 D BLE     : subscribe complete, ep = 0x7303eb6780
```

### Platform

android

### Platform Version(s)

1.0

### Type

Manually tested with SDK, Hardware validated

### Anything else?

Logs:

[meross-failure-btsnoop_hci.log](https://github.com/project-chip/connectedhomeip/files/10876231/meross-failure-btsnoop_hci.log)

[meross-failure-export.txt](https://github.com/project-chip/connectedhomeip/files/10876232/meross-failure-export.txt)

[image]

bzbarsky-apple, 2023-03-06:
> Initially the spec specified that the BLE characteristics should operate using BLE Notifications, but was changed to specify that BLE Indications should be used. 

No, initially the spec said to use Indications.  Then there was an attempt to change to Notifications, that attempt failed to get uptake and actually become certifiable, as I understand it, and it was reverted.

bzbarsky-apple, 2023-03-06:
> https://github.com/project-chip/connectedhomeip/pull/19051

And that change was months before SVE.  How did something from before that change get certified?

bzbarsky-apple, 2023-03-06:
In particular, the device being described would not, as I understand it, work with _any_ 1.0-SDK-based controller.  Why are we fixing one specific controller to work with such a device?

@andy31415 @woody-apple

chrisdecenzo, 2023-03-06:
Please feel free to reach out to me or James for specific details on this. The problem has impacted many platforms since not all picked up the change in June, and I believe there are devices on the market with firmware today which do not have the change. I don't believe iOS commissioners were impacted, but Android commissioners which use the SDK's android port are impacted. The problem has easily caused hundreds of hours of dev/debug time to date, hence the motivation for this change which is intended to make commissioners backward compatible with devices which do not have the fix.

andy31415, 2023-03-06:
@chrisdecenzo could you provide some specific examples here? We could fix the debugging issue without introducing non-spec behavior if that is all that we would prioritize. So questions:

- why is iOS behavior different than android? Did their usage of indication vs notification happen at different times?
- are there actual certified devices that are non-spec compliant here? Or are these test devices or something else?

A change that landed in June and release finished in November seems like sufficient time for things to get picked up. There are a lot of other deltas between june and november which likely would need to be picked up.

andy31415, 2023-03-06:
For `there are devices on the market with firmware today which do not have the change` ... do we have explicit examples (at least one)? Are they actually matter certified (wondering how if this is non-spec compliant)?

PR #25444 (2023-03-03): [Android] Fallback to BLE Notification when Indication unsupported
To support devices based on older revisions of the SDK that did not include the GATT Server Characteristic property change (from Notification to Indication), the GATT Characteristic properties are queried to determine if Indication and/or Notification is supported. When available, Indication is preferred.

Fixes #25442 

Example logs from device supporting only Notifications:

```
6.654011000; ATT Write Transaction (Client Characteristic Configuration: Notifications=Enabled, Indications=Disabled)                        ; 01 00; [... cut]