project-chip/connectedhomeip #25452: [BUG] M5 crashed when establishing CASE

kpark-apple, 2023-03-03:
### Reproduction steps

1. Commission M5 device
2. Open commissioning window
3. Commission another fabric

M5 crashed after receiving Sigma3.

```
I (38635) chip[SC]: Received Sigma3 msg
E (38645) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
E (39735) chip[DL]: Failed to post event to CHIP Platform event queue
E (39735) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (39795) chip[DL]: Failed to post event to CHIP Platform event queue
E (39795) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (39815) chip[DL]: Failed to post event to CHIP Platform event queue
E (39815) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (40055) chip[EM]: <<< [E:24304r M:32978080 (Ack:57566611)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
I (40065) chip[IN]: (U) Sending msg 32978080 to IP address 'UDP:[FE80::44:449F:A55E:6644%st1]:57882'
I (40075) chip[SC]: SecureSession[0x3ffc8968]: Moving from state 'kEstablishing' --> 'kActive'
I (40075) chip[IN]: CASE Session established to peer: <573108A51DE43D73, 2>
I (40085) chip[IN]: CASE Server enabling CASE session setups
E (40095) chip[DL]: Long dispatch time: 1494 ms, for event type 3
E (40065) chip[DL]: Failed to post event to CHIP Platform event queue

abort() was called at PC 0x4014066b on core 0

Backtrace: 0x40081e82:0x3ffdf790 0x400926fd:0x3ffdf7b0 0x40098fb2:0x3ffdf7d0 0x4014066b:0x3ffdf840 0x401d397d:0x3ffdf8a0 0x401d3a78:0x3ffdf8f0 0x40095add:0x3ffdf910
```

[2023-02-21 at 05:14:31 PM.txt](https://github.com/project-chip/connectedhomeip/files/10878078/2023-02-21.at.05.14.31.PM.txt)

### Bug prevalence

Observed once

### GitHub hash of the SDK that was being used

df4ad9521b1dac98d7965f455fec74ffa270429c

### Platform

esp32

shubhamdp, 2023-03-08:
Sorry Boris it's pretty hard to guess the call stack by just looking at these logs.

@kpark-apple its highly unlikely that you would have kept the `.elf` file for this build, but if you happen to have one that would help to debug that call stack.

I'll try to reproduce this at my end on the same SHA and see if I hit the same crash.

dhrishi, 2024-10-25:
Ideally this shoulld be fixed with the M5 stack changes for memory. Please re-open if seen