project-chip/connectedhomeip #25472: [BUG] Unable to dim light on Google Home App with Matter bridge app

yurenlee, 2023-03-04:
### Reproduction steps

Based on the examples/bridge-app reference using the v1.0.0.2 tag in the Matter SDK, I have added a dimmable light using the OnOff Cluster and LevelControl Cluster. It works fine in the Apple Home app, but in the Google Home app, I can only turn it on or off. The dimmable button is disabled by default, and the dimming feature does not work without sending any interaction commands.

PS:
Matter SDK Version: v1.0.0.2
Google Home App Version: 2.65.24.1-dogfood
Google Nest Display Version: 1.56.324896

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

v1.0.0.2

### Platform

esp32

bzbarsky-apple, 2023-03-06:
@yurenlee A few things:

1. Generally, bugs about specific controllers (e.g. Google Home) should be reported to those controllers, not here.
2. It's hard to say anything intelligent about this situation without knowing what the endpoints defined in your application look like: What device types they expose, what clusters they expose, etc.

closed 2023-03-06 by bzbarsky-apple

yurenlee, 2023-03-06:
@bzbarsky-apple 

Thanks, I have posted the same issue on the Google Nest developer community. However, I am confused about the mandatory clusters of the dimmable light device type for the bridge app. There are no certain examples for reference.

bzbarsky-apple, 2023-03-06:
A dimmable light device requires Identify, Groups, Scenes (but not in practice, since Scenes is provisional), On/Off, and Level Control servers.