project-chip/connectedhomeip #25504: [Feature] We'll need a way to programmatically/dynamically generate certs for various VID/PID (VendorID/ ProductID) for extended testing with ota-requester app.

vinay-apple, 2023-03-07:
### Feature description

...
SHA: df4ad9521b1dac98d7965f455fec74ffa270429c

We'll need a way to programmatically/dynamically generate certs for various VID/PID (VendorID/ ProductID) for extended testing with ota-requester app. 

We're getting the vid/pid from the certificate now. The test certificate used is hardcoded to have a VendorID of 0xFFF1. 

So if we try to change the value at below location to 0xFFF0 (#define CHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID 0xFFF0) for testing this would still use the hardcoded value of 0xFFF1 on the compiled certificate on the ota-requester app hence we get unexpected results on the testing. 

https://github.com/project-chip/connectedhomeip/blob/a7cc2cf3ca94693e0ba08578cc1365d4848890dd/src/include/platform/CHIPDeviceConfig.h#L200 

[ota-requester-hard-coded-VID-0xFFF1.txt](https://github.com/project-chip/connectedhomeip/files/10904132/ota-requester-hard-coded-VID-0xFFF1.txt)

### Platform

darwin

### Platform Version(s)

13.2

bzbarsky-apple, 2023-03-07:
@vinay-apple We already have different example DA certs for different product ids.  Would simply adding some for a few different vendor ids be sufficient here?  What set of (VID, PID) pairs would you want to be supported?

bzbarsky-apple, 2023-03-08:
OK, so generating new DAC/PAI stuff is doable.  The hard part is the Certification Declaration.  That is also hardcoded to VID=0xFFF1, and it needs to be signed by one of the alliance private keys, so generating a new one is somewhat complicated.

@tcarmelveilleux @emargolis is there some way we could also get CDs for VIDs 0xFFF2 and 0xFFF3 generated?

Also of note: the CDs for VID=0xFFF1 in `src/app/tests/suites/credentials/TestHarnessDACProvider.cpp` and `src/credentials/examples/DeviceAttestationCredsExample.cpp` are different, but both seem to be signed by the alliance?

closed 2023-03-09 by bzbarsky-apple