project-chip/connectedhomeip #25508: [BUG] Failed to build cryptography

olalonde, 2023-03-07:
### Reproduction steps

I clone the repo and ran `source scripts/bootstrap.sh` on MacOS M1. 

Relevant logs:

```
  build/temp.macosx-10.14.6-arm64-cpython-38/cryptography.hazmat.bindings._openssl.c:590:10: fatal error: 'openssl/opensslv.h' file not found
  #include <openssl/opensslv.h>
           ^~~~~~~~~~~~~~~~~~~~
  1 error generated.
  /private/var/folders/h_/17w4t45s5bdgm84gjtrj2vjm0000gn/T/pip-build-env-ie70p3oe/overlay/lib/python3.8/site-packages/setuptools/command/build_py.py:202: SetuptoolsDeprecationWarning:     Installing 'cryptography.hazmat.bindings._rust' as data is deprecated, please list it in `packages`.
      !!
```

[bootstrap-log.txt](https://github.com/project-chip/connectedhomeip/files/10905146/bootstrap-log.txt)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

a7cc2cf3ca94693e0ba08578cc1365d4848890dd

### Platform

darwin

### Platform Version(s)

Darwin mbair 21.4.0 Darwin Kernel Version 21.4.0: Mon Feb 21 20:36:53 PST 2022; root:xnu-8020.101.4~2/RELEASE_ARM64_T8101 arm64

olalonde, 2023-03-07:
```
brew link openssl --force
```

seems to fix the issue

bzbarsky-apple, 2023-03-07:
That's pretty odd.  @olalonde Why is bootstrap trying to compile rust stuff for you?

olalonde, 2023-03-07:
I'm not sure. I do have Rust installed. I just tried bootstrapping again with a fresh repo checkout and hitting this bug again.

bzbarsky-apple, 2023-03-07:
Do you have some sort of environment variables set that would affect what pigweed is doing?  Because the other two bugs you filed also indicate that something very weird is going on with your bootstrap....

For what it's worth, I just tried doing a bootstrap without openssl linked (but installed, so that homebrew python can do https) on an M1 in a clean checkout and it worked fine.

olalonde, 2023-03-08:
I can't see anything special. Here are my environment variables:

[env.txt](https://github.com/project-chip/connectedhomeip/files/10917486/env.txt)

I also tried to `brew unlink openssl` and  check out from scratch again but still getting that openssl error.

bzbarsky-apple, 2023-03-08:
@mspang @kcoppock any idea what might be going on here?

olalonde, 2023-03-08:
Not sure if that helps...

```console
$ rustup show                                                                                                                                                                                  master
Default host: aarch64-apple-darwin
rustup home:  /Users/redacted/.rustup

installed toolchains
--------------------

stable-aarch64-apple-darwin
nightly-2022-05-15-aarch64-apple-darwin
nightly-2022-10-25-aarch64-apple-darwin
nightly-aarch64-apple-darwin (default)
stage1

installed targets for active toolchain
--------------------------------------

aarch64-apple-darwin
thumbv7em-none-eabihf

active toolchain
----------------

nightly-aarch64-apple-darwin (default)
rustc 1.68.0-nightly (333ee6c46 2023-01-18)

1. ```

olalonde, 2023-03-08:
I tried to `brew install python` and `brew link --overwrite python@3.11` and bootstrap seemed to work now.