project-chip/connectedhomeip #25509: [BUG] ModuleNotFoundError: No module named 'click'

olalonde, 2023-03-07:
### Reproduction steps

There was an error when running the script first time (I forgot to set `ANDROID_NDK_HOME`). I set it and tried running the script again but now am just getting `ModuleNotFoundError: No module named 'click'`. 

```console
$ ./scripts/build/build_examples.py --target android-arm64-chip-tool build   
2023-03-06 23:38:59 INFO    Building targets: android-arm64-chip-tool
2023-03-06 23:38:59 INFO    Preparing builder 'android-arm64-chip-tool'
2023-03-06 23:38:59 INFO    Generating /Users/redacted/code/connectedhomeip/out/android-arm64-chip-tool
2023-03-06 23:38:59 INFO    Setting up Android deps through Gradle
2023-03-06 23:39:00 INFO    Downloading https://services.gradle.org/distributions/gradle-7.2-bin.zip
2023-03-06 23:39:23 INFO    ..........10%...........20%...........30%...........40%...........50%...........60%...........70%...........80%...........90%...........100%
2023-03-06 23:39:24 INFO
2023-03-06 23:39:24 INFO    Welcome to Gradle 7.2!
2023-03-06 23:39:24 INFO
2023-03-06 23:39:24 INFO    Here are the highlights of this release:
2023-03-06 23:39:24 INFO     - Toolchain support for Scala
2023-03-06 23:39:24 INFO     - More cache hits when Java source files have platform-specific line endings
2023-03-06 23:39:24 INFO     - More resilient remote HTTP build cache behavior
2023-03-06 23:39:24 INFO
2023-03-06 23:39:24 INFO    For more details see https://docs.gradle.org/7.2/release-notes.html
2023-03-06 23:39:24 INFO
2023-03-06 23:39:24 INFO    Starting a Gradle Daemon (subsequent builds will be faster)
2023-03-06 23:39:26 INFO    > Task :buildSrc:compileJava NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:compileGroovy
2023-03-06 23:39:34 INFO    > Task :buildSrc:processResources NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:classes
2023-03-06 23:39:34 INFO    > Task :buildSrc:jar
2023-03-06 23:39:34 INFO    > Task :buildSrc:assemble
2023-03-06 23:39:34 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2023-03-06 23:39:34 INFO    > Task :buildSrc:test NO-SOURCE
2023-03-06 23:39:34 INFO    > Task :buildSrc:check UP-TO-DATE
2023-03-06 23:39:34 INFO    > Task :buildSrc:build
2023-03-06 23:39:35 INFO    > Task :copyArtifacts
2023-03-06 23:39:36 INFO    > Task :setUpAndroidDeps
2023-03-06 23:39:36 INFO
2023-03-06 23:39:36 INFO    BUILD SUCCESSFUL in 35s
2023-03-06 23:39:36 INFO    4 actionable tasks: 4 executed
2023-03-06 23:39:36 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2023-03-06 23:39:36 INFO    Setting up Java deps
2023-03-06 23:39:39 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
Traceback (most recent call last):
  File "./scripts/build/build_examples.py", line 221, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/redacted/code/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "./scripts/build/build_examples.py", line 211, in cmd_build
    context.obj.Build()
  File "/Users/redacted/code/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/Users/redacted/code/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/Users/redacted/code/connectedhomeip/scripts/build/builders/android.py", line 329, in generate
    self.validate_build_environment()
  File "/Users/redacted/code/connectedhomeip/scripts/build/builders/android.py", line 140, in validate_build_environment
    raise Exception(
Exception: Environment ANDROID_NDK_HOME missing, cannot build android libraries
~/code/connectedhomeip$ vim ~/.zshrc.local; source ~/.zshrc.local  
~/code/connectedhomeip0 ./scripts/build/build_examples.py --target android-arm64-chip-tool build  
Traceback (most recent call last):
  File "./scripts/build/build_examples.py", line 21, in <module>
    import click
ModuleNotFoundError: No module named 'click'
```
 
Maybe because the build partially completed? Is there a `clean` command I can use and try again?

### Bug prevalence

whenever i do this

### GitHub hash of the SDK that was being used

a7cc2cf3ca94693e0ba08578cc1365d4848890dd 

### Platform

darwin

### Platform Version(s)

Darwin mbair 21.4.0 Darwin Kernel Version 21.4.0: Mon Feb 21 20:36:53 PST 2022; root:xnu-8020.101.4~2/RELEASE_ARM64_T8101 arm64

olalonde, 2023-03-07:
I tried with `--clean` but still same problem:

```console
0 ./scripts/build/build_examples.py --target android-arm64-chip-tool --clean                                                                                                                   master
Traceback (most recent call last):
  File "./scripts/build/build_examples.py", line 21, in <module>
    import click
ModuleNotFoundError: No module named 'click'
```

bzbarsky-apple, 2023-03-07:
@andy31415