project-chip/connectedhomeip #25510: [BUG] "command not found: pw"

olalonde, 2023-03-07:
### Reproduction steps

```console
~/code/connectedhomeip0 source ./scripts/bootstrap.sh                                                                                                                                                                master
Synchronizing submodule url for 'examples/common/QRCode/repo'
Synchronizing submodule url for 'examples/common/m5stack-tft/repo'
Synchronizing submodule url for 'third_party/boringssl/repo/src'
Synchronizing submodule url for 'third_party/bouffalolab/repo'
Synchronizing submodule url for 'third_party/cirque/repo'
Synchronizing submodule url for 'third_party/editline/repo'
Synchronizing submodule url for 'third_party/freertos/repo'
Synchronizing submodule url for 'third_party/imgui/repo'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/30739A0'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/CYW930739M2EVB-01'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/btsdk-include'
Synchronizing submodule url for 'third_party/infineon/cyw30739_sdk/repos/btsdk-tools'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/TARGET_CY8CKIT-062S2-43012'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/abstraction-rtos'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/anycloud-ota'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/bluetooth-freertos'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/btstack'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/clib-support'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/connectivity-utilities'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/core-lib'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/core-make'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/freertos'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/kv-store'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/lwip'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/mbedtls'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/mtb-hal-cat1'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/mtb-pdl-cat1'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/psoc6cm0p'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/recipe-make-cat1a'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/retarget-io'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/secure-sockets'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/serial-flash'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/whd-bsp-integration'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/wifi-connection-manager'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/wifi-host-driver'
Synchronizing submodule url for 'third_party/infineon/psoc6/psoc6_sdk/libs/wifi-mw-core'
Synchronizing submodule url for 'third_party/jsoncpp/repo'
Synchronizing submodule url for 'third_party/libwebsockets/repo'
Synchronizing submodule url for 'third_party/mbed-mcu-boot/repo'
Synchronizing submodule url for 'third_party/mbed-os-cypress-capsense-button/repo'
Synchronizing submodule url for 'third_party/mbed-os-posix-socket/repo'
Synchronizing submodule url for 'third_party/mbed-os/repo'
Synchronizing submodule url for 'third_party/mbedtls/repo'
Synchronizing submodule url for 'third_party/mt793x_sdk/filogic'
Synchronizing submodule url for 'third_party/mt793x_sdk/lwip'
Synchronizing submodule url for 'third_party/mt793x_sdk/mDNSResponder'
Synchronizing submodule url for 'third_party/nanopb/repo'
Synchronizing submodule url for 'third_party/nlassert/repo'
Synchronizing submodule url for 'third_party/nlfaultinjection/repo'
Synchronizing submodule url for 'third_party/nlio/repo'
Synchronizing submodule url for 'third_party/nlunit-test/repo'
Synchronizing submodule url for 'third_party/nxp/mw320_sdk/repo'
Synchronizing submodule url for 'third_party/open-iot-sdk/sdk'
Synchronizing submodule url for 'third_party/open-iot-sdk/storage'
Synchronizing submodule url for 'third_party/openthread/ot-efr32'
Synchronizing submodule url for 'third_party/openthread/ot-ifx'
Synchronizing submodule url for 'third_party/openthread/ot-nxp'
Synchronizing submodule url for 'third_party/openthread/ot-qorvo'
Synchronizing submodule url for 'third_party/openthread/repo'
Synchronizing submodule url for 'third_party/ot-br-posix/repo'
Synchronizing submodule url for 'third_party/pigweed/repo'
Synchronizing submodule url for 'third_party/pybind11/repo'
Synchronizing submodule url for 'third_party/qpg_sdk/repo'
Synchronizing submodule url for 'third_party/silabs/gecko_sdk'
Synchronizing submodule url for 'third_party/silabs/matter_support'
Synchronizing submodule url for 'third_party/silabs/wiseconnect-wifi-bt-sdk'
Synchronizing submodule url for 'third_party/simw-top-mini/repo'
Synchronizing submodule url for 'third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx'
Synchronizing submodule url for 'third_party/ti_simplelink_sdk/repo_cc32xx'

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (4.9s)
  Setting up Python environment.....done (44.3s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

/Users/redacted/code/connectedhomeip/.environment/activate.sh:202: command not found: pw
Error during bootstrap--see messages above.
```

### Bug prevalence

whenever i do this

### GitHub hash of the SDK that was being used

a7cc2cf3ca94693e0ba08578cc1365d4848890dd 

### Platform

darwin

### Platform Version(s)

Darwin mbair 21.4.0 Darwin Kernel Version 21.4.0: Mon Feb 21 20:36:53 PST 2022; root:xnu-8020.101.4~2/RELEASE_ARM64_T8101 arm64

bzbarsky-apple, 2023-03-07:
@andy31415

y-morgunov, 2023-03-21:
Once I was helped by deleting .environment #17193 
Another time deleting /home/vscode/pigweed helped
So far, I have not figured out what is causing this problem.