project-chip/connectedhomeip #25516: Window Covering Level and Slat control on dynamic Bridge on Homekit

mustafayuce33, 2023-03-07:
### Reproduction steps

I use dynamic bridge to publish window covering devices to platforms such as Homekit, Google Home, Alexa etc. I finished Lightbulb and Switch types. Now I work on window covering and I face with some problems.

The problem is that I do not see any slider for both lift and tilt positioning. The only thing that I can do is On-Off

I also setup some attribute Ids in window covering such as `ZCL_WC_END_PRODUCT_TYPE_ATTRIBUTE_ID` with all combinations, but still no success. 

Is there anything else that I should do to see Lift and Tilt positioning as slider on `Homekit`?

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

2ce5f2d6e88a69e24ca8542cd03c233292755423

### Platform

raspi

### Type

Platform Issue

bzbarsky-apple, 2023-03-07:
You should probably check the feature map for your cluster: does it have the right bits set?

bzbarsky-apple, 2023-03-07:
But if the feature map is right, @mustafayuce33, you probably want to file a feedback assistant ticket and report the ticket id here.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

mustafayuce33, 2023-03-07:
@bzbarsky-apple What should I set the feature map? I did a couple of things but nothing worked

bzbarsky-apple, 2023-03-07:
> What should I set the feature map?

What features does your cluster implement?

> So if you can help that would be great.

Just to be clear: I might be able to help if the issue is in your app, but I don't work on Home and if the issue is in Home I will not be able to help with that, apart from pointing people to the feedback assistant ticket.

mustafayuce33, 2023-03-07:
@bzbarsky-apple  I set feature map as `0x001F` to enable all feature that I found here. May be that was wrong?

```
// Bitmap for Feature
enum class Feature : uint32_t
{
    kLift              = 0x1,
    kTilt              = 0x2,
    kPositionAwareLift = 0x4,
    kAbsolutePosition  = 0x8,
    kPositionAwareTilt = 0x10,
};
```

bzbarsky-apple, 2023-03-07:
That looks ok, then.  In that case, the feedback assistant ticket is the way to go.

bzbarsky-apple, 2023-03-07:
I have no idea whether Home examines those and if so what it looks for.

mustafayuce33, 2023-03-07:
So only featureMap is the attribute ID that describe whether device has tilt or lift characteristic right?

bzbarsky-apple, 2023-03-07:
Per spec, yes.  What clients do is up to them.

woody-apple, 2023-06-23:
Looks like the feedback ticket has been filed, closing as this is a request for Home to a feature.

closed 2023-06-23 by woody-apple