project-chip/connectedhomeip #25532: [BUG] TC_RR_1_1 group sub-test not run if group cluster endpoint is not 0

lucicop, 2023-03-07:
### Reproduction steps

- have group cluster configured on an endpoint different than 0
- run TC_RR_1_1.py test - will skip group cluster tests, due to no ACL access (ACL admin privileges are always on endpoint 0)

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

b89e83be43dde7a79d90823f4bc1279eb53f76de

### Platform

python

bzbarsky-apple, 2023-03-07:
> As it is not mandatory to have the group cluster on endpoint 0

@lucicop Are you talking about the _Groups_ cluster or the _Group Key Management_ cluster here?  The latter is in fact required to be on endpoint 0.

lucicop, 2023-03-08:
I edited the title to indicate more clearly that I am referring to Groups cluster. It is on endpoint 0 in lighting-common zap configuration, as far as I can tell.

lucicop, 2023-03-08:
Suggest to replace ACL entry with this one (if Groups cluster is defined to be on endpoint 1):
`admin_targets = [
            Clusters.AccessControl.Structs.Target(endpoint=0),
            Clusters.AccessControl.Structs.Target(endpoint=1),
            Clusters.AccessControl.Structs.Target(cluster=0xFFF1_FC00, deviceType=0xFFF1_BC30)
        ]` 

More structural changes need to be added to dynamically determine the Groups cluster endpoint and use that in the ACL entry.

bzbarsky-apple, 2023-03-08:
> It is on both endpoint 0 and 1 in lighting-common zap configuration.

That's extremely broken.  :(

tehampson, 2023-03-09:
Naive question, but why do you think the test is not run. From the code I see:
```
        logging.info("Step 10: Validating groups support minimums")
        groups_cluster_endpoints: Dict[int, Any] = await dev_ctrl.ReadAttribute(self.dut_node_id, [Clusters.Groups])
        counted_groups_clusters: int = len(groups_cluster_endpoints)
        # The test for Step 10 and all of Steps 11 to 14 are only performed if Groups cluster instances are found.
        if counted_groups_clusters > 0:
            ....
```

That read attribute does not specify any endpoint and is a wildcard for the endpoint.

lucicop, 2023-03-09:
In theory this should be the case - but for reasons unclear to me, due to the ACL entries giving administer privileges on endpoint 0, Groups cluster on endpoint 1 is not reported, unless ACL entries are extended to give admin privileges to EP1 as well.

tehampson, 2023-03-09:
Okay I just got a crash course on ACLs. This is my understanding of the issue from the test side:

In test step we are "Add 4 Access Control entries on DUT with a list of 4 Subjects and 3 Targets". We are doing this because this is the minimum matter devices are expected to support per fabric. We cannot add another endpoint as a target as then we would be require something above the minimum. When we do the read attribute in step 10, we don't satisfy the Subject of any of the other Access Control entries so we can only focus on modifying the Admin Access Control entry. For the admin we have a current target:
```
        # - Targets field: [
        #                   {Endpoint: 0},
        #                   {Cluster: 0xFFF1_FC00, DeviceType: 0xFFF1_FC30},
        #                   {Cluster: 0xFFF1_FC00, DeviceType: 0xFFF1_FC31}
        #                 ]
```

We could possibly change `{Cluster: 0xFFF1_FC00, DeviceType: 0xFFF1_FC30},` to `{Cluster: 0x0000_0004},` This would still fulfill the minimum number of targets, while allowing the test step 10 to actually do the read it needs to do. The only thing I am not 100% sure about is if doing something like this would mean we are technically testing for slightly lower memory requirements. I don't see any other options right now

tehampson, 2023-03-09:
I am Yak shaving quite a bit for this issue. In updating ACL to allow it to find all Groups clusters [(rough draft PR with extra debugging stuff)](https://github.com/project-chip/connectedhomeip/pull/25596) this uncovered a real problem where indicated_max_groups_per_fabric is incorrect provided by DUT was not right. In trying to fix that by updating `CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC`, I am now seeing a weird race like condition where in step 14 when the test is trying to issue a readAttribute command, for some reason the app DUT running on host seems to be hung and doesn't reply but after the MRP is done retrying it then receives the read attribute request, but it is already too late to respond. I have no idea why this is happening

tehampson, 2023-03-13:
Quick update `ReadGroupTable` in `src/app/clusters/group-key-mgmt-server/group-key-mgmt-server.cpp` is taking 7.8s to complete on a host machine. To fix this github issue we should probably break this down into 4 separate smaller issues each with their own PRs, and likely should go in the following order:
* Identify why `ReadGroupTable` take so long when there are something like 60 groups, otherwise all clusters app will not pass CI, and fix
* Identify why written entry endpoints doesn't match the read entry endpoints in the `validate_group_table`, and fix
* Update example `CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC` to match how many Groups clusters there are for the various examples
* Update ACL in TC_RR_1_1 so that all Groups cluster on endpoints other than 0 can be discovered

tehampson, 2023-03-15:
I am incrementally working on this issue (roughly 1 hour per day). If there is a rush for this issue to be resolved please let me know as currently I am treating this with a fairly low priority.

Update for today:
* Timing issue:
  * `ReadGroupTable` calls `encoder.Encode` 60 times. This is because there are 5 fabrics, each fabric has 3 Group endpoints that support the max number of groups which is 4.  5*3*4 = 60. Because of fabric filtering, the time taken to perform `encoder.Encode` is roughly 580ms for each the 12 table entries that we are encoding and 0ms for the other 48.
  * `GroupTableCodec::EncodeForRead` is where we have to figure out the break down of that 580ms as that is what is consuming.
    * Inside the `while (iter->Next(mapping))` [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/group-key-mgmt-server/group-key-mgmt-server.cpp#L90):
      * This is happening 36 times for this particular RR_1_1 case. I actually don't full understand why this is 36 for the time being
      * The while loop itself is reporting almost no time at all, but interestingly it is the `iter->Next(mapping))` that is taking on average 16ms. 16*36 = 576ms which is roughly most of the time spend in each instances in this loop. Next up I will dig into why `iter->Next(mapping))` is taking up so much time.
* `validate_group_table`. Talking to Chris, who originally wrote that section of the test mentioned that there is likely an issue with how we are adding group in `add_all_groups`. He was trying to write unique group IDs across the different endpoints. He will fix this particular portion of this issue

tehampson, 2023-03-16:
Today's power hour update:
* Timing issue:
  * All of the time that creates this issue happens in to every `Load` from persistent memory in `GroupDataProviderImpl::EndpointIteratorImpl::Next` over [here](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/GroupDataProviderImpl.cpp#L1263). This `Load` operation cost 6ms to 7ms because of how full the persistent memory is.
  * The 36 comes from another bug exposed by the issue `add_all_groups` that needs to be fixed. We are adding more groups then the fabric is reported support as a max. I will open a separate issue for that as I do not think it is critical in fixing this specific issue.
    - [ ]  Create a github issue for this specific task
  * With a local hack to `EndpointIteratorImpl` I can specify a target group ID that we are interested in so that the iterator is more focused. This cuts down the number of iteration from 36 to 3 for a single run [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/group-key-mgmt-server/group-key-mgmt-server.cpp#L90). Once Chris make a fix to `add_all_groups` this will further be reduced it from 3 calls to just 1. This will bring down the time spent from roughly 576ms to just 16ms.
    * I will formalize my current hack to something that is actually submit-able and push a PR on my next working day.

tehampson, 2023-03-21:
Turns out calling `AddGroup` on `CHIP_CONFIG_MAX_GROUPS_PER_FABRIC * number_of_endpoints_with_groups_cluster` is what we should be doing and that the test spec is not correct (Side note I learned that `AddGroup` is not adding a group, it is adding an endpoint to a group). I have filed https://github.com/CHIP-Specifications/chip-test-plans/issues/2502. If that is test spec issue filed is correct then https://github.com/project-chip/connectedhomeip/pull/25767 will be abandoned and we will update `add_all_groups` to add all endpoints to all 12 groups (which it is actually currently doing, just that the data it builds for validation in `validate_group_table` needs to be corrected to give all the endpoints in the list.

cjandhyala, 2023-03-22:
@tehampson since this is one of the critical TC we want matter devices to support, Could you pls expedite the fix for this script, 'd like to get this ready before our final 1.1 TH release ready.

tehampson, 2023-03-22:
The current status is there is some discussion being had about what is written in the spec and what is the intent. The spec says `The node SHALL support at least four groups per fabric per endpoint having a Groups cluster instance`. How this is understood is that `CHIP_CONFIG_MAX_GROUPS_PER_FABRIC` would need to be `4 * num_of_endpoint_instances_of_group_cluster`. Some are arguing that the intent should be that `CHIP_CONFIG_MAX_GROUPS_PER_FABRIC` should still be allowed to be 4, and that we just need to make sure that there is enough space for all endpoints to be able to join 4 group.

If we want to follow the spec I'll need to get these two PRs approved as they currently are:
* https://github.com/project-chip/connectedhomeip/pull/25789
* https://github.com/CHIP-Specifications/chip-test-plans/pull/2503

If we need to go by the presumed intent then we need to ingrate the suggested changes in https://github.com/CHIP-Specifications/chip-test-plans/issues/2504 into https://github.com/CHIP-Specifications/chip-test-plans/pull/2503 (very quick), and I need to update https://github.com/project-chip/connectedhomeip/pull/25789 which would also be quick.

My hands are tied right now because I don't know which of the two I need to be making sure the test is written for. How it is written in the spec or the presumed intent

@bzbarsky-apple any ideas who needs to be involve to clear this up any quicker?

bzbarsky-apple, 2023-03-23:
> Some are arguing that the intent should be that CHIP_CONFIG_MAX_GROUPS_PER_FABRIC should still be allowed to be 4

Just to be clear, I am _not_ arguing that.  The intent is clearly that devices should support `4 * num_of_endpoint_instances_of_group_cluster` groups, so that each endpoint should be able to be in 4 groups with no two endpoints being in the same group.  

The open question is how many endpoints per group devices should be required to support (e.g. whether each endpoint is required to be able to be placed in all `4 * num_of_endpoint_instances_of_group_cluster` groups) and what this test should test about that, if anything.

In terms of who is involved...  I believe Chris and Tennessee wrote the spec bits, but Tennessee does not have bandwidth for this right now.  Probably good to check with Leo and maybe Rob Houtepen?

tehampson, 2023-03-23:
Okay so I think I was misunderstanding the intended functionality that Boris was referring to. Now I get it what we need to do to move forward. If we are all in agreement

* This Issue is not correct https://github.com/CHIP-Specifications/chip-test-plans/issues/2502 and should just be closed as it is currently written as intended
* This issue is not correct https://github.com/CHIP-Specifications/chip-test-plans/issues/2504 and should just be closed as it is currently written as intended
* We do need this test spec issues https://github.com/CHIP-Specifications/chip-test-plans/issues/2505 addressed this and it does need to be incorporated.

If there is any disagreement please let me know, otherwise this is what I am working towards and will update
https://github.com/CHIP-Specifications/chip-test-plans/pull/2503 to reflect only updating ACL, and I will update https://github.com/project-chip/connectedhomeip/pull/25789 to contain all the required fixes to get RR_1_1 passing

tehampson, 2023-03-24:
Both [test spec](https://github.com/CHIP-Specifications/chip-test-plans/pull/2503) the two SDK changes are merged https://github.com/project-chip/connectedhomeip/pull/25758 and https://github.com/project-chip/connectedhomeip/pull/25789

I think you should be good to go and test this now

closed 2023-03-24 by tehampson