project-chip/connectedhomeip #25533: [BUG] TC_RR_1_1 ACL test might not work as intended

lucicop, 2023-03-07:
### Reproduction steps

While running the ACL write/read steps of TC_RR_1_1, there is a caching of ACL entries, so a read-back will not result in reading from KVS storage (as might have been intended for the test ?), therefore this will not validate the correct storage of ACL entries in KVS.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

b89e83be43dde7a79d90823f4bc1279eb53f76de

### Platform

python

bzbarsky-apple, 2023-03-07:
Should the test include a reboot step somewhere to test the persistance aspect of things?

bzbarsky-apple, 2023-03-07:
@tcarmelveilleux @cecille

cecille, 2023-03-07:
I think I might be misunderstanding the bug - where is the caching concern? On the device or on the TH? Which steps?

lucicop, 2023-03-08:
I think the idea behind ACL write/read-back is to also validate the correct KVS storage of the data. I notice no read-back from KVS during ACL steps, so I concluded there must be a caching mechanism behind this data. In this case, the read-back will not provide much value, if not validating the correct KVS storage, which I think is more important to be validated.

cecille, 2023-03-08:
I don't think the write / read steps need to be changed here - the spec doesn't mandate that attributes immediately be pushed through to long-term storage as long as they persist across reboot. How the device internally handles persistance is up to the device and is therefore not certifiable. 
 
Testing for correct persistence behavior on reboot in general might be worthwhile, but I that would apply to all attributes that have the Persistent quality. That might be a good test to add, but I think it might be easier to do as a separate test.

The other possible concern here is that that devices might have sufficient memory to hold the incoming data, but not sufficient storage. A reboot when the device is at the max might be worthwhile to test that in theory, but in practice, getting a device into a low-storage situation is not going to be something we can reliably write a cert test for. It's just too dependent on the implementation, especially since many devices use flash so have fairly complex schemes.

For ACLs specifically, the more interesting potential issue is the partial write. But again, that's a really hard test to force as a cert test.

tehampson, 2023-03-09:
>That might be a good test to add, but I think it might be easier to do as a separate test.

I would agree with Cecille, [step 3 from the test spec](Looking at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/minimal_resource_reqs.adoc) says "Add 4 Access Control entries on DUT". The python script step 3a is writing those values and 3b is validating that the added ACLs before proceeding with the rest of the test.

cjandhyala, 2023-09-19:
closing the issue stale. Pls re-open if the issue still needs to be addressed

closed 2023-09-19 by cjandhyala