project-chip/connectedhomeip #25566: [BUG] ESP lighting build failed

haihuynhDF, 2023-03-08:
### Reproduction steps

I used: IDF 5.0.1, MacOS, chip M1
1. Run the command
```
idf.py build
```
2. Got the error
```
[1/1] cd /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/bu...nectedhomeip/examples/lighting-app/esp32/build/bootloader/bootloader.binBootloader binary size 0x6740 bytes. 0x8c0 bytes (8%) free.
[257/344] Performing build step for 'chip_gn'FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip/lib/libCHIP.a
cd /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/chip && ninja esp32
[566/690] c++ obj/third_party/connectedhomeip/third_party/nlfaultinjection/repo/src/libnlfaultinjection.nlfaultinjection.cpp.oFAILED: obj/third_party/connectedhomeip/third_party/nlfaultinjection/repo/src/libnlfaultinjection.nlfaultinjection.cpp.o
/Users/df/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -MMD -MF obj/third_party/connectedhomeip/third_party/nlfaultinjection/repo/src/libnlfaultinjection.nlfaultinjection.cpp.o.d -Og -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/config -isystem/Users/df/matter/esp-idf/components/freertos/include/freertos -isystem/Users/df/matter/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -isystem/Users/df/matter/esp-idf/components/newlib/platform_include -isystem/Users/df/matter/esp-idf/components/freertos/FreeRTOS-Kernel/include -isystem/Users/df/matter/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -isystem/Users/df/matter/esp-idf/components/freertos/esp_additions/include/freertos -isystem/Users/df/matter/esp-idf/components/freertos/esp_additions/include -isystem/Users/df/matter/esp-idf/components/esp_hw_support/include -isystem/Users/df/matter/esp-idf/components/esp_hw_support/include/soc -isystem/Users/df/matter/esp-idf/components/esp_hw_support/include/soc/esp32 -isystem/Users/df/matter/esp-idf/components/esp_hw_support/port/esp32/. -isystem/Users/df/matter/esp-idf/components/esp_hw_support/port/esp32/private_include -isystem/Users/df/matter/esp-idf/components/heap/include -isystem/Users/df/matter/esp-idf/components/log/include -isystem/Users/df/matter/esp-idf/components/soc/include -isystem/Users/df/matter/esp-idf/components/soc/esp32 -isystem/Users/df/matter/esp-idf/components/soc/esp32/include -isystem/Users/df/matter/esp-idf/components/hal/esp32/include -isystem/Users/df/matter/esp-idf/components/hal/include -isystem/Users/df/matter/esp-idf/components/hal/platform_port/include -isystem/Users/df/matter/esp-idf/components/esp_rom/include -isystem/Users/df/matter/esp-idf/components/esp_rom/include/esp32 -isystem/Users/df/matter/esp-idf/components/esp_rom/esp32 -isystem/Users/df/matter/esp-idf/components/esp_common/include -isystem/Users/df/matter/esp-idf/components/esp_system/include -isystem/Users/df/matter/esp-idf/components/esp_system/port/soc -isystem/Users/df/matter/esp-idf/components/esp_system/port/include/private -isystem/Users/df/matter/esp-idf/components/xtensa/include -isystem/Users/df/matter/esp-idf/components/xtensa/esp32/include -isystem/Users/df/matter/esp-idf/components/lwip/include -isystem/Users/df/matter/esp-idf/components/lwip/include/apps -isystem/Users/df/matter/esp-idf/components/lwip/include/apps/sntp -isystem/Users/df/matter/esp-idf/components/lwip/lwip/src/include -isystem/Users/df/matter/esp-idf/components/lwip/port/include -isystem/Users/df/matter/esp-idf/components/lwip/port/freertos/include -isystem/Users/df/matter/esp-idf/components/lwip/port/esp32xx/include -isystem/Users/df/matter/esp-idf/components/lwip/port/esp32xx/include/arch -isystem/Users/df/matter/esp-idf/components/bt/include/esp32/include -isystem/Users/df/matter/esp-idf/components/bt/common/osi/include -isystem/Users/df/matter/esp-idf/components/bt/common/api/include/api -isystem/Users/df/matter/esp-idf/components/bt/common/btc/profile/esp/blufi/include -isystem/Users/df/matter/esp-idf/components/bt/common/btc/profile/esp/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -isystem/Users/df/matter/esp-idf/components/bt/porting/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/port/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/transport/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/nimble/nimble/include -isystem/Users/df/matter/esp-idf/components/bt/host/nimble/esp-hci/include -isystem/Users/df/matter/esp-idf/components/esp_timer/include -isystem/Users/df/matter/esp-idf/components/esp_wifi/include -isystem/Users/df/matter/esp-idf/components/esp_event/include -isystem/Users/df/matter/esp-idf/components/esp_phy/include -isystem/Users/df/matter/esp-idf/components/esp_phy/esp32/include -isystem/Users/df/matter/esp-idf/components/esp_netif/include -isystem/Users/df/matter/esp-idf/components/mbedtls/port/include -isystem/Users/df/matter/esp-idf/components/mbedtls/mbedtls/include -isystem/Users/df/matter/esp-idf/components/mbedtls/mbedtls/library -isystem/Users/df/matter/esp-idf/components/mbedtls/esp_crt_bundle/include -isystem/Users/df/matter/esp-idf/components/fatfs/diskio -isystem/Users/df/matter/esp-idf/components/fatfs/vfs -isystem/Users/df/matter/esp-idf/components/fatfs/src -isystem/Users/df/matter/esp-idf/components/wear_levelling/include -isystem/Users/df/matter/esp-idf/components/esp_partition/include -isystem/Users/df/matter/esp-idf/components/sdmmc/include -isystem/Users/df/matter/esp-idf/components/driver/include -isystem/Users/df/matter/esp-idf/components/driver/deprecated -isystem/Users/df/matter/esp-idf/components/driver/dac/include -isystem/Users/df/matter/esp-idf/components/driver/gpio/include -isystem/Users/df/matter/esp-idf/components/driver/gptimer/include -isystem/Users/df/matter/esp-idf/components/driver/i2c/include -isystem/Users/df/matter/esp-idf/components/driver/i2s/include -isystem/Users/df/matter/esp-idf/components/driver/ledc/include -isystem/Users/df/matter/esp-idf/components/driver/mcpwm/include -isystem/Users/df/matter/esp-idf/components/driver/pcnt/include -isystem/Users/df/matter/esp-idf/components/driver/rmt/include -isystem/Users/df/matter/esp-idf/components/driver/sdio_slave/include -isystem/Users/df/matter/esp-idf/components/ [... cut ...]
[575/690] c++ obj/third_party/connectedhomeip/src/transport/tests/libTransportLayerTests.TestSessionManager.cpp.oninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stderr_output_35392 and /Users/df/matter/connectedhomeip/examples/lighting-app/esp32/build/log/idf_py_stdout_output_35392
```

### Bug prevalence

When ever I do it

### GitHub hash of the SDK that was being used

6bb89174aaf3ee57c06c21fee11bc9c911c55f07

### Platform

esp32

bzbarsky-apple, 2023-03-08:
@haihuynhDF Did this use to work, or is this the first time you are trying it?

@shubhamdp @dhrishi

shubhamdp, 2023-03-08:
@haihuynhDF IDF 5.0.1 is not tested and there may be possibility that build will fail. Latest supported IDF version is [v5.0](https://github.com/espressif/esp-idf/releases/tag/v5.0)

ivishaltejwani, 2023-03-16:
I am able to build it using idf 5.0.1 @shubhamdp  , 

unable to add in apple home or google home stuck on 

ble con(0)

do I need , hubs like homepod or google home to add?

dhrishi, 2024-10-25:
Currently the ESP32 lighting app is building fine (executed in CI). Please try with the latest supported ESP-IDF. Closing this.

closed 2024-10-25 by dhrishi