project-chip/connectedhomeip #25568: [BUG] scripts/bootstrap.sh

mmt-10, 2023-03-08:
### Reproduction steps

```bash
cd ${MATTER-SDK}
source ./scripts/bootstrap.sh
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 536a0d9b0f738ed79a939c638ef1bbfae3e7cf59 (HEAD -> master, origin/master, origin/HEAD)

### Platform

other

### Anything else?

## Environment

OS: ubuntu 20.04

python --version : Python 3.8.10

## Log

I think brew is for Mac.

> 
>   BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
> 
> Downloading and installing packages into local source directory:
> 
>   Setting up CIPD package manager...[|]======================================================================
> It looks like this version of Python does not support SSL. This is common
> when using Homebrew. If using Homebrew please run the following commands.
> If not using Homebrew check how your version of Python was built.
> 
> brew install openssl  # Probably already installed, but good to confirm.
> brew uninstall python && brew install python
> ======================================================================                                                                                                                     Traceback (most recent call last):
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
>     sys.exit(main())
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
>     return EnvSetup(**vars(parse())).setup()
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
>     result = step(spin)
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 595, in cipd
>     cipd_client = cipd_wrapper.init(
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 378, in init
>     bootstrap(client, silent, rosetta=rosetta)
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 328, in bootstrap
>     tmp.write(client_bytes(rosetta))
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 228, in client_bytes
>     conn = https_connect_with_proxy(CIPD_HOST)
>   File "/home/muto/matter/matter-master-20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 201, in https_connect_with_proxy
>     conn = httplib.HTTPSConnection(url.hostname, url.port)
>   File "/usr/lib/python3.8/http/client.py", line 1383, in __init__
>     super(HTTPSConnection, self).__init__(host, port, timeout,
>   File "/usr/lib/python3.8/http/client.py", line 834, in __init__
>     (self.host, self.port) = self._get_hostport(host, port)
>   File "/usr/lib/python3.8/http/client.py", line 868, in _get_hostport
>     i = host.rfind(':')
> AttributeError: 'NoneType' object has no attribute 'rfind'

bzbarsky-apple, 2023-03-08:
@mmt-10 Your python install on Ubuntu is somehow configured without SSL/TLS support.  Did you just install it from packages (which ones?) or did you do something manual to get that python?

mmt-10, 2023-03-09:
Thank you for your reply.
I checked my PC, and PC may be wrong.

closed 2023-03-09 by mmt-10

aaaaa654, 2023-03-09:
Hey guys, I got a problem with bootstrap.sh as well, CPID package was fine but the python environment is causing problem, i'm not a developer so it would be great if anyone can have a look and help me, thanks
[image]
[image]

bzbarsky-apple, 2023-03-09:
@aaaaa654 Do you have the devel packages for glib and gio-unix installed?  In particular, did you follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#prerequisites where is says to install various packages including `libglib2.0-dev`?

aaaaa654, 2023-03-09:
> @aaaaa654 Do you have the devel packages for glib and gio-unix installed? In particular, did you follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#prerequisites where is says to install various packages including `libglib2.0-dev`?

yes i did install those packages and followed all the steps, this is the only problem i encountered in the whole process.

bzbarsky-apple, 2023-03-09:
Well, your pkg-config does not seem to be aware of the libglib2.0-dev bits... what does running:
```
pkg-config --cflags --libs glib-2.0
```
at the shell show you?

aaaaa654, 2023-03-10:
It says the pkg isn't found in the path
[image]

bzbarsky-apple, 2023-03-10:
OK, do you have a `glib-2.0.pc` file somewhere on your disk? If so where?  And what, if anything, is your `PKG_CONFIG_PATH` environment variable set to?

bzbarsky-apple, 2023-03-10:
I'd start with seeing whether `locate glib-2.0.pc` says anything useful, @aaaaa654

LiuHuanYin, 2023-07-24:
> 20230308/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 228, in client_bytes
> conn = https_connect_with_proxy(CIPD_HOST)
> File [...]

Please check the https_proxy settings. 
Do not work: `https_proxy=127.0.0.1:7890`
Works well: `https_proxy=http://127.0.0.1:7890`