project-chip/connectedhomeip #25581: Re-arm the fail-safe for a longer period to deal with CASE retries

bzbarsky-apple, 2023-03-08:
There are several retry mechanisms in CASE session setup at this point:

1. There is a certain amount of retrying during DNS-SD resolution.  This can take tens of seconds.
2. Once we have a resolution result, there are retries across the multiple IPs we got back.  Each one can take at least 9 seconds, and possibly longer depending on the advertised MRP parameters.
3. Once we run out of IPs, there are the retries we do to handle devices that are somewhat slow to connect to the operational network.  This can repeat all of the above a few times, with some backoff.

Looks like CHIP_DEVICE_CONFIG_FAILSAFE_EXPIRY_LENGTH_SEC defaults to 60 and I suspect no one is changing it in practice, and 60 seconds is just not going to be enough to go through that (plus the rest of commissioning)

Some possible options here:

1) AutoCommissioner could automatically extend the fail-safe by some amount before trying operational discovery.  It already does this if you pass in a value for `SetCASEFailsafeTimerSeconds` on the commissioning parameters, but we could have it default to some nonzero value.  A concern there would be overriding a longer fail-safe that someone else set up before triggering the AutoCommissioner.  And there is no way I see to read back the current fail-safe value from the server so we don't override it with a shorter one.
2) We could change things so that failures in CASE establishment trigger an extension of the fail-safe.  Again, how long is a good question, since we don't want to make the fail-safe _shorter_.  This would also make OperationalSessionSetup a little bigger, since it would now need a callback for "going to retry".
3) We could leave the existing setup and punt the problem to API consumers who would need to call `SetCASEFailsafeTimerSeconds` on the commissioning parameters.  This solves the "we're shortening their fail-safe" problem, but leaves the problem of them having to guess how long to extend for without knowing ahead of time how many times CASE will actually be tried.  That said, they could set this to something that corresponds to their "if we have not commissioned by then, we give up" timeout.
4) We could do something where we pipe through the "going to retry" notification all the way to the commissioning client, so they can decide what to do.  This obviously takes a bit more surgery.
5) What else?

bzbarsky-apple, 2023-03-09:
https://github.com/project-chip/connectedhomeip/pull/25595 implements option 2.  I figured if we extend by "how long will a retry take" plus some overhead for actually doing the CASE handshake and sending the CommissioningComplete command, that would be pretty good as a start.

closed 2023-03-10 by bzbarsky-apple

ivishaltejwani, 2023-03-17:
assuming this would be needed in espmatter too , should be updated there